Bodmer/TFT_eSPI #730: ST7796S disappearing image

gilphilbert, 2020-08-17:
I bought this display, based on the ST7796S (SPI variant). The board appears to be a straight passthrough, with SPI coming from the controller itself:
[https://www.aliexpress.com/item/4000572980706.html?spm=a2g0s.9042311.0.0.531b4c4dt1eWTo](url)
It's a 4" device with a 480x320 resolution.

It's connected to the Espressif DevKit (ESP32) and I configured the device with the correct dimensions and driver (`#define DRIVER_ST7796`) and specified all pins including the backlight pin.

When running code (any of the examples do the same) the display shows the drawn image for a fraction of a second then the image is instantly erased and replaced with a blank screen. If I use one of the examples with a 2s delay in the loop, the image is displayed just long enough to see it's there, then it instantly disappears. This happens every two seconds. I've tried different power supplies but the effect is the same.

Any suggestions?

Bodmer, 2020-08-17:
This is the setup and pins I use for that display:
```

#define ST7796_DRIVER

#define TFT_CS   5  // Chip select control pin
#define TFT_DC   26 // Data Command control pin
#define TFT_RST  27 // Reset pin

#define TOUCH_CS 22     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

 #define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS

#define SPI_READ_FREQUENCY  16000000

#define SPI_TOUCH_FREQUENCY  2500000

```

Make sure you select the ESP32 Dev Module option (not Wrover) in the boards menu.

Provide more information on your hardware and software environment if that does not help.

gilphilbert, 2020-08-17:
Well it's definitely faster, but it's still flickering!

In order to get it to do anything, I had to add lines for SPI (MISO, MOSI and SCK).

**User_Setup.h**
```
#define ST7796_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5  // Chip select control pin
#define TFT_DC   26 // Data Command control pin
#define TFT_RST  27 // Reset pin

#define TOUCH_CS 22     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

 #define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS

#define SPI_READ_FREQUENCY  16000000

#define SPI_TOUCH_FREQUENCY  2500000
```
I have the screen wired as the comments show. It's a 5V unit, so it's connected to the 5V pin on the unit. I've tried an external power supply but that made no difference. I'm using PlatformIO, but I'm using the library pulled from GitHub rather than the one in PlatformIO (it's out of date). Nothing else is connected to the board. The touch pins are currently not connected (irritatingly, it has separate SPI pins for the touch panel...) so only the TFT pins are connected. The SD card is not connected.

I'm happy to provide any other detail, just let me know what you need.

closed 2020-09-17 by Bodmer