Bodmer/TFT_eSPI #731: reading GSCAN (45h) of ST7796S

svofski, 2020-08-19:
This is not an issue with TFT_eSPI as such, more like a point of research that might result in a useful feature. Please excuse me if I chose the wrong place to ask.

I'm on ESP32 WROVER-B with a 480x320 ST7796S display, the usual red board variety from Ali. For all the regular things it seems to be working fine at 80MHz (which is btw a point of confusion as the datasheet lists clock periods that suggest 15Mhz at most). 

I'm trying to sync with the refresh to avoid tearing. For this purpose,  GSCAN register (0x45) seems to be suitable -- it returns current scan line. But whenever I touch that register, the display seems to stop scanning completely. 

Here's how I'm trying to do that. I've just added a method to TFT_eSPI:
```
uint16_t TFT_eSPI::read_gscan()
{
    writecommand(0x45); // e.g. 0xda would do nothing bad
    begin_tft_read();
    DC_D;
    dummy8 = tft_Read_8();
    uint8_t hi = tft_Read_8();
    uint8_t lo = tft_Read_8();
    end_tft_read();
    return (hi << 8) | lo;
}
```
If I call this at least once, the display just shows blank screen. Everything continues operating as usual, just the display is blank white. I tried accessing different registers, not all of them are destructive. Trying to read RDID1 register does not do any harm for example (although I'm not sure if it returns anything meaningful, but that's another topic).

Maybe I'm not doing it right? Could you suggest something? Thanks a lot!

Here's the datasheet that I'm looking at (Page 196, 9.2.36 GSCAN (45h): Get Scanline):

https://cdn.hackaday.io/files/1625926956336128/ST7796S.pdf

svofski, 2020-08-23:
Here's the version that seems to be working for me, it gives me what seems to be reasonable line numbers (0..483).

```
uint16_t TFT_eSPI::read_gscan()
{
    begin_tft_read();
    DC_C;
    tft_Write_9(0x45);  // add a dummy bit
    DC_D;
    uint8_t hi = tft_Read_8();
    uint8_t lo = tft_Read_8();
    dummy8 = tft_Read_8(); // dummy8 is a volatile uint8_t;
    end_tft_read();
    return (hi << 8) | lo;
}
```

Now I really cannot call this a "solution" because it matches nothing in the datasheet. The dummy read is supposed to be the first byte, but I have to do it in the end. It seems to be a silly thing to do at all, but if I don't include the dummy read, the next time the values will be wrong. The datasheet seems to show D/C bit as a part of SPI transaction, while here it's clearly an extra line. This confuses me too.

Bodmer, 2020-09-07:
Thanks for posting the code.  I have also run tests and agree that the data sheet appears to be wrong based on empirical evidence. This is my test code which gives the expected results for the scan line and is essentially the same as yours:
```
#define tft_Write_9(C) TFT_WRITE_BITS(C, 9)

uint16_t TFT_eSPI::read_gscan()
{
    begin_tft_read();
    DC_C;
    tft_Write_9(0x45);  // Add extra clock
    DC_D;
    uint16_t scanline = (tft_Read_8()<<8) | tft_Read_8();
    tft_Read_8();       // Dummy read needed
    end_tft_read();
    return scanline;
}
```

It also appears that with the ST7796 and SPI interface you have to wait 4ms after any register read otherwise the next write to the display is ignored.

closed 2020-09-07 by Bodmer