Bodmer/TFT_eSPI #736: Memory freed twice when using 4-bit sprites

jpasqua, 2020-08-23:
When using 4-bit sprites, the following sequence leads to memory being freed twice (and consequent error):

````
createSprite
createPalette
deleteSprite
createSprite
createPalette <- error occurs here in call to free()
````

The issue is that in deleteSprite, if `_colorMap != nullptr`, it is freed; however, it is not set to `nullptr`. On a later call to `createPalette`, it notes that `_colorMap != nullptr` and tries to free it again.

Here is what I did to fix the problem in my local copy:

````
void TFT_eSprite::deleteSprite(void)
{
  if (!_created ) return;

  if (_colorMap != nullptr)
  {
    free(_colorMap);
    _colorMap = nullptr; // <---- Make sure to reset this to _nullptr
  }

  free(_img8_1);

  _img8 = nullptr;

  _created = false;
}
````

Bodmer, 2020-08-26:
Thanks for pointing this out. I will fix it soon but I am currently busy on urgent an project.

jpasqua, 2020-08-26:
Thank you. No rush on my account.

commit 731228d1 (2020-08-28): Fix deleteSprite

Fixes #736 and also delete color map even if the sprite has not been created.
closed 2020-08-28 by Bodmer