Bodmer/TFT_eSPI #737: Touch screen keypad with esp32 and ble

justdrago, 2020-08-26:
Requesting help I am trying to integrate esp32 ble with touchscreen keypad i am currently using a matrix keypad to perform the task but wanted to change it below is what i currently am using and when i hit the D key on the 4x4 matrix it sends hello world to whatever text application i am using on my computer using blue tooth any help would be greatly appreciated

void loop() {
  char key = keypad.getKey();

  // Only do something with the keypress if we
  // are connected to something via bluetooth
  
  if (bleKeyboard.isConnected() && key) {
    Serial.println(key);
    switch (key) {
      case 'D':
        Serial.println("Sending 'Hello world'...");
        bleKeyboard.print("Hello world");
        break;
}

    delay(100);
    bleKeyboard.releaseAll(); // this releases the buttons
  }
}

Bodmer, 2020-08-26:
I am not familiar with that keypad hardware so am unable to help. I can only answer questions related to the TFT library. Get the TFT_eSPI examples running with the touch screen and you should be able to copy the code.

justdrago, 2020-08-27:
Hi, Ok thank you I didn't know if you would know or maybe someone else that has used this library. Since the keypad in question is the one in the example i have those working as intended. I just havent figured out how to make it behave like a regular keypad so if i press a button i can make it print it to a text application like note pad

closed 2020-08-28 by Bodmer