Bodmer/TFT_eSPI #738: Sprite behaving strange

videobuff, 2020-08-26:
I have boith a 2.8 and 3.5 inch dispay with either the ILI9341 or the  ILI9488. Both work with touch as well.
I like to combine the animated dial and the keypad, but i didn't get that far. With the animated dial (with both screens and libraries)  the dial is eating up the numbers. The video shows clearly what i mean. I haven't been able to solve the issue so far. Latest library installed again today. Help is much appreaciated......

Erik

Here is the YouTube video showing the issue
[https://youtu.be/l1MuyuVZ7D8]

Bodmer, 2020-08-26:
The screen CGRAM must be readable for the animated dial code to work, so you have to connect TFT SDO. This should work fine with the ILI9341.  The ILI9488 does not tristate the SDO line so add a 1K Ohm series resistor at the TFT SDO pin end of the connection.

videobuff, 2020-08-26:
Thanks very much. It works. Now i will combine the animated dial in compass lay-out together with the keypad to control a hamradio rotor. The screen works, now the functions. I really like your library a lot.
Erik, PA0ESH
[image]

closed 2020-08-28 by Bodmer