Bodmer/TFT_eSPI #739: Imagen pixelate

kondorcl, 2020-09-03:
Please can someone help me, I am using SPI ST7796 320x480 display and esp32_sdcard_jpeg example with JPEGDecoder.h  V1.8.0. And I get the image pixelated. With another display ST7735 and appropiated user_setup work fine.
TFT_eSPI ver = 2.2.19
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7796
Display width  = 320
Display height = 480

Another problem. If update JPEGDecoder to 1.8.1 i have error:no matching function for call to 'min(uint16_t&, uint32_t)'

Regards,
ERamirez

Bodmer, 2020-09-07:
To solve the "min" problem change the lines to:
```
  uint32_t min_w = jpg_min(mcu_w, max_x % mcu_w);
  uint32_t min_h = jpg_min(mcu_h, max_y % mcu_h);
```
I will update the example.

Not sure what is causing the missing pixels (or is it missing MCU blocks?). I will create a tet setup and investigate.

kondorcl, 2020-09-07:
Mr. Bodmer thanks very much for reading my query.
I updated JPEGDecoder library to 1.8.1 with your recommendation change min(...) to jpg_min(....), works ok; but, the  missing pixels go on. In this site you can watch the problem : [http://avandes1600.xyz/st7796/Baboon.png]. I changed USB feed to external power supply and I have the same result.

Regards,
ERamirez

Bodmer, 2020-09-07:
Does the JPEGDecoder SPIFFS_Jpeg example work when line 125 is enabled thus:
```
drawJpeg("/Baboon40.jpg", 0, 0); // 320 x 480 image
```

kondorcl, 2020-09-07:
Yes sir, and below the printed lines and I 'm using SDcard.

===========================
Drawing file: /Baboon40.jpg
===========================
JPEG image info
===============
Width      :320
Height     :480
Components :1
MCU / row  :20
MCU / col  :30
Scan type  :4
MCU width  :16
MCU height :16
===============

  JPEG drawn in 487 ms

Regards,
ERamirez

Bodmer, 2020-09-08:
The picture you provided a link to indicates that all the data is being read correctly from the SD Card (otherwise the image decode would fail) but occasionally the blocks are not getting sent to the display for some reason.

What SPI clock rate are you using for the TFT?
Run the "Read_User_Setup" diagnostic example and post the results.

kondorcl, 2020-09-08:
Bodmer, here is: Display SPI frequency = 27.00
I changed SPI Freq.  to:
SPI_FREQUENCY  40000000
SPI_FREQUENCY  20000000
SPI_FREQUENCY   1000000 
and get the same test result.

Thanks again

Regards,
ERamirez

Bodmer, 2020-10-31:
Unfortunatley I cannot reporoduce the problem so am unable to advise further.

closed 2020-10-31 by Bodmer