Bodmer/TFT_eSPI #740: Ambiguous function signature

QrackEE, 2020-09-05:
Hi,

I just got involved in project which uses this library, so I don't know much about it yet, but the issue seems strictly related to API change on update. In the project we had version 2.2.12 or thereabouts of the library and after updating to latest (v2.2.19) it no longer compiled due to ambiguous `img.pushSprite(0,0,TFT_TRANSPARENT);` calls. Simply enough, the compiler can not decide on specific function:
```
void     pushSprite(int32_t x, int32_t y, uint16_t transparent);
```
versus introduced in the update
```
bool     pushSprite(TFT_eSprite *spr, int32_t x, int32_t y);
```
My solution was to change out calls to `img.pushSprite(0,0,(uint16_t)TFT_TRANSPARENT);`, which works, but is easy to forget and not user-friendly. My proposition is to use *reference* instead of *pointer* and change new function signature to:
```
bool     pushSprite(TFT_eSprite &spr, int32_t x, int32_t y);
```
and implementation accordingly. If there are any reasons why pointer is preferred, I hope this post help those who may stumble upon this issue.

Bodmer, 2020-09-06:
Which processor board are you using?

QrackEE, 2020-09-06:
I have custom board with ESP32-PICO-D4. Do you think it might be toolchain-specific? To be honest, I didn't think of that, but it may be the case...

Bodmer, 2020-09-06:
I will check this as I did test the function but maybe did not use the variable types that show the error. Do you have a simple example sketch that shows the error?

QrackEE, 2020-09-07:
No, I don't have an a simple sketch at hand. This should be reproducible with function call from original post ('img.pushSprite(0,0,TFT_TRANSPARENT);'). I can prepare it and try,bit not earlier than in the afternoon here (let's say 10h from now :D)

Bodmer, 2020-09-07:
Yep, you are right. That call does produce an ambiguous error. I did not see this as I used explicit varaibles in the call.

Since there is no example yet for the " bool     pushSprite(TFT_eSprite *spr, int32_t x, int32_t y);" I thin it is reasonable to accept your push request. Thanks.

Bodmer, 2020-09-07:
I have decided to revert this change as it introduces an inconsistency.

I will think on this further but a function name change seems the better way forward since this function in new and currently has no example.

QrackEE, 2020-09-07:
Glad to help. Regarding the fix, I'm also not sure what is better approach, to change argument list (as proposed) or change the name - thought you'd prefer to keep the name. I agree that my proposition introduces inconsistency, so let's try to find better name: *embedSprite*?

Bodmer, 2020-09-07:
New function name is pushToSprite, this name is consistent with other functions where the sprite is pushed somewhere else.

Library updated.

commit d7fdcc09 (2020-09-07): Fix #740 and #704

#740: To avoid ambiguity the pushSprite for writing a sprite to another sprite has been renamed pushToSprite

#704: Change to 18 bit colurs for SSD1963 with SPI interface

Remove outdated TFT_SPIFFS_Jpeg example
closed 2020-09-07 by Bodmer