Bodmer/TFT_eSPI #742: Cyrillic symbol on screen

PavelAravin, 2020-09-06:
Hello. I try to print Cyrillic on screen with TFT_eSPI library but doesn't work.
Using [this font](https://github.com/immortalserg/AdafruitGFXRusFonts/blob/master/FontsRus/TimesNRCyr8.h) and println function. If use write(hex), for example tft.write(0xAF) i see corresponding symbol "Я" or 0xAE - "Ю" etc, it is right, if using tft.print or println symbol not printing, for example tft.println("Библиотека для TFT") on screen render only "TFT". How can this problem be solved?

Bodmer, 2020-09-06:
This is because the library by default assumes UTF-8 encoding for characters outside the normal ASCII range 32-127.

To switch off the UTF-8 decoding use the following command:
```
tft.setAttribute(UTF8_SWITCH, false);
```
The print strings should then work. Report back if it does or does not work.

PavelAravin, 2020-09-07:
Yes, it work. Thanks for feedback and good library. 
[image]

closed 2020-09-07 by PavelAravin