Bodmer/TFT_eSPI #746: Wrong colors

Tinkerpete, 2020-09-07:
I used the library last year and all work good!

Now i have a problem with the color from a sprite, which is defined so:

// 'Moon', 50x50px
const uint16_t PROGMEM Moon [] = {
0x0000, 0x0000, 0x0000, 0x0000, 0x1020, 0x2020, 0x3040, 0x3840, 0x3040, 0x3040, 0x3040, 0x2820, 0x2020, 0x1820, 0x1000, ...

I have read the hint with '.setSwapBytes(true) or .setSwapBytes(false)' but this did not change anything. Have you an idea?

In my code i use this so:

TFT_eSprite moonSprite = TFT_eSprite(&tft);
...
moonSprite.createSprite(51, 51);
moonSprite.fillSprite(TFT_BLACK); // For transparency
moonSprite.pushImage(1, 1, 50, 50, Moon);
...
skySprite.fillRect(0, 0, SCREENWIDTH, SCREENHEIGHT / 2, TFT_BLACK);
skySprite.pushImage(animationXPos, animationYPos, 50, 50, Moon);
skySprite.pushSprite(0, 0);
...

Bodmer, 2020-09-07:
setSwapBytes() can be used with the tft or sprite class:

Make sure you are using the right one. e,g,
tft.setSwapBytes(true)  means swap bytes written to TFT
moonSprite.setSwapBytes(true) means swap bytes in image written to moonSprite
skySprite.setSwapBytes(true)  means swap bytes in image written to skySprite

If that does not help provide a simplified but complete example that illustrates the problem.

Tinkerpete, 2020-09-07:
Thank you! This was the right command:

skySprite.setSwapBytes(true);

closed 2020-09-08 by Bodmer