Bodmer/TFT_eSPI #747: Use modified glcd font

PavelAravin, 2020-09-11:
Dear Bodomer, i try find method to display cyrrilic with fill background, i found on internet modified glcd font for adafruit library with support cyrrilic symbols and text backgroung, but adafruit library less stable.
How can connect it with TFT_eSPI? I replace file glcdfont.c and comment all stroke related with font in user_setting except first stroke #define LOAD_GLCD  and compile this code 
 tft.init();
  tft.setTextFont(1);
  tft.setAttribute(UTF8_SWITCH, false);
  tft.fillScreen(TFT_WHITE);
  tft.setCursor(0, 0, 4);
   tft.setTextColor(TFT_RED, TFT_BLACK);
   tft.println("Привет, hello");

but it no work. 
what am I doing wrong?

Bodmer, 2020-09-11:
See #742 

I assume you can display other Latin ASCII characters (32-127)?

PavelAravin, 2020-09-11:
No display, clear white screen, if remove comment slash next to font2,
font3, 4, 5,6,7,8 then Latin symbol is visible, Cyrrilic none

Bodmer, 2020-09-11:
To use background you must use the drawString() functions, this is because the print class sends 1 character at a time and since the character bounding boxes may overlap (e.g. itallic fonts) a simple algorithms to determine the blanking area does not work well.

PavelAravin, 2020-09-11:
I understand, but this is an alternative to the standard font, namely its replacement. There is no slope, and the symbols are specially made as simple as possible. It's weird that it doesn't work.

Bodmer, 2020-09-11:
Provide a copy of the font file and I will see if it is in a compatible format.

Bodmer, 2020-09-11:
Note that:

tft.setTextFont(0);

Selects the glcd font.

PavelAravin, 2020-09-12:
Yes! It work. I use this glcd.c file [glcdfont.txt](https://github.com/Bodmer/TFT_eSPI/files/5212175/glcdfont.txt)  and utf8rus function.

```
String utf8rus(String source)
{
  int i,k;
  String target;
  unsigned char n;
  char m[2] = { '0', '\0' };

  k = source.length(); i = 0;

  while (i < k) {
    n = source[i]; i++;

    if (n >= 0xC0) {
      switch (n) {
        case 0xD0: {
          n = source[i]; i++;
          if (n == 0x81) { n = 0xA8; break; }
          if (n >= 0x90 && n <= 0xBF) n = n + 0x30;
          break;
        }
        case 0xD1: {
          n = source[i]; i++;
          if (n == 0x91) { n = 0xB8; break; }
          if (n >= 0x80 && n <= 0x8F) n = n + 0x70;
          break;
        }
      }
    }
    m[0] = n; target = target + String(m);
  }
return target;
}
```
Example: 
tft.setTextColor(TFT_WHITE, TFT_BLACK);
tft.setTextFont(0);
tft.setTextSize(2);
tft.setAttribute(UTF8_SWITCH, false);
tft.printl(utf8rus("Мама мыла Милу мылом"));

Symbols displayed with worked background. May be this way to make any language character set?

Showed up1 bug or not bug. The setTextFont parameter must be specified in each function separately, before println.
If it's not difficult, describe an example of using padding in order not to use fillRect if the line has become shorter.
[image]

[image]

Bodmer, 2020-09-12:
If I understand correctly you have solved it now.

Great!

closed 2020-09-12 by Bodmer