Bodmer/TFT_eSPI #748: shared SmoothFont between multiple sprites

svofski, 2020-09-12:
I need to draw many text scrollers on the screen. I use Smooth font feature, the font data is from a SPIFFS file on ESP32.
When the text needs to change, update is taking a lot of time. Part of the problem seems to be the requirement to loadFont() for every sprite. I use the same font for everything, so naturally the idea is to share the same font instead of loading/unloading it for every sprite. Is this possible? Thanks!

Bodmer, 2020-09-12:
It is not possible with the current library implementation. I would put the smooth font in a FLASH array (program memory) this is the same memory that SPIFFS uses but you do not have the overhead of the filing system and buffering. Maybe that will get better performance.

Anpther option is to just have one sprite for the font rendering.  Not that you can now render a window area of a Sprite to screen, so you can have different size scrollers all from one sprite.

If you have a full example of what you are trying to achieve then I can look at it and advise further.

svofski, 2020-09-12:
Thanks for the clarification! This was my idea after a bit of studying the code, but I wanted to be sure that I'm not overlooking something. Using a Flash array sounds like a good idea, I'll try that. Using a single sprite for all text rendering will not work in my case because font drawing is very slow compared to blitting a sprite. 

As a hopeful solution I'm working on pooling the sprites, this should minimise the amount of text drawing. 

The project is a train timetable. Many repetitive texts, but some of them don't fit in one line so they need to be scrolled. With all sprites scrolling I get full screen refresh in 25ms. When I have to redraw the text, it's like 0.5seconds if not worse.

Bodmer, 2020-09-13:
A further option is to have a sprite, render all the text into it, then copy that sprite into another one and copy that to the screen. If using a SPI display then you can DMA the sprite to the screen while creating the next one.

svofski, 2020-09-13:
I have implemented a pool and this reduced the amount of redrawing the text to a minimum. The idea of using a text-rendering prototype sprite is very interesting though and I think I might have a use for it. Thank you so much for your help!

closed 2020-09-13 by Bodmer