Bodmer/TFT_eSPI #751: Transparency is lost when using pushRotated

devbender, 2020-09-14:
Hi Bodmer,

First of all thanks for all your work on this great library. You have made me question what I thought was not possible with an MCU and a cheap screen.

I tried using both pushRotated methods to push a sprite with transparent color and none of them seem to work. The sprite is rotated with no problem but the transparent color is lost. Not sure if I might be missing something. 

Hardware: M5stack-Core-Esp32
TFT_eSPI ver: 2.2.19

`bool     pushRotated(int16_t angle, int32_t transp = -1);
bool     pushRotated(TFT_eSprite *spr, int16_t angle, int32_t transp = -1);`

I didn't see any examples using these so I created one to illustrate:

`#include <TFT_eSPI.h>

#define centerX tft.width()/2
#define centerY tft.height()/2

#define widthX  tft.width()
#define heightY tft.height()

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sp1 = TFT_eSprite(&tft);
TFT_eSprite sp2 = TFT_eSprite(&tft);

void setup() {
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setPivot(centerX, centerY);

  // Sprite 1
  sp1.setColorDepth(8);  
  sp1.createSprite(320, 240);
  sp1.fillSprite(0x02B5);
  
  // Sprite 2
  sp2.setColorDepth(16);  
  sp2.createSprite(100, 100);
  sp2.fillSprite(TFT_TRANSPARENT);
  sp2.fillRect(10, 10, 40, 40, TFT_RED);

  // CASE 1: Push sprite1 then sprite2 with transparency >> WORKS OK!
  sp1.pushSprite(0,0);
  sp2.pushSprite(centerX-50, centerY-50, TFT_TRANSPARENT);

  delay(2000); tft.fillScreen(TFT_BLACK);

  // CASE 2: Push sprite1 then rotated sprite2 with transparency >> TRANSP IS LOST  
  sp1.pushSprite(0,0);
  sp2.pushRotated(30, TFT_TRANSPARENT);

  delay(2000); tft.fillScreen(TFT_BLACK);

  // CASE 3: Push rotated sprite2 to sprite1 then push sprite1 to tft >> TRANSP IS LOST
  sp2.pushRotated(&sp1, 45, TFT_TRANSPARENT);  
  sp1.pushSprite(0,0);
}

void loop() {}`

Bodmer, 2020-09-14:
The problem is lack of memory. To use such large sprites you need an ESP32 board with PSRAM.

e.g. 320 x 240 8 bit sprite needs 76,800 bytes
      100 x 100 16 bit sprite needs 20,000 bytes
Total 96,800 bytes

The ESP32 starts to have problems when too much RAM is allocated, usually it crashes as there is not enough work space left.

Options:
1. Use smaller sprites
2. Use a board with PSRAM and enable it in the IDE
3. Delete sprites to free up memory (note this may lead to memory fragmentation and an eventual crash after many hours.

The ESP32 only reserves about 90kbytes for program use.

devbender, 2020-09-14:
I tried the same example using the M5Stack-Fire which does have PSRAM and got the same results.

See example bellow using a smaller sprite as background and detailing memory usage.

```
#include <TFT_eSPI.h>

#define centerX tft.width()/2
#define centerY tft.height()/2

#define widthX  tft.width()
#define heightY tft.height()

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sp1 = TFT_eSprite(&tft);
TFT_eSprite sp2 = TFT_eSprite(&tft);

void setup() {
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setPivot(centerX, centerY);

  uint32_t heapStart = ESP.getFreeHeap();
  uint32_t psramStart = ESP.getFreePsram();

  Serial.begin(115200);  
  Serial.println("\n>>>> ESP-STARTED <<<<\n");
  Serial.printf("> Internal Total heap %d kb, internal Free Heap %d \n", ESP.getHeapSize(), ESP.getFreeHeap());
  Serial.printf("> SPIRam Total heap %d kb, SPIRam Free Heap %d \n", ESP.getPsramSize(), ESP.getFreePsram()); 
  Serial.printf("> ChipRevision %d, Cpu Freq %d, SDK Version %s\n",ESP.getChipRevision(), ESP.getCpuFreqMHz(), ESP.getSdkVersion());
  Serial.printf("> Flash Size %d kb, Flash Speed %d Hz\n",ESP.getFlashChipSize(), ESP.getFlashChipSpeed());

#if defined(BOARD_HAS_PSRAM)
  Serial.printf("*** BOARD HAS PSRAM ***\n");
#else
  Serial.printf("*** BOARD DOES NOIT HAVE PSRAM ***\n");
#endif

  // Sprite 1
  sp1.setColorDepth(8);  
  sp1.createSprite(200, 200);
  sp1.fillSprite(0x02B5);
  
  // Sprite 2
  sp2.setColorDepth(16);  
  sp2.createSprite(100, 100);
  sp2.fillSprite(TFT_TRANSPARENT);
  sp2.fillRect(10, 10, 40, 40, TFT_RED);

  // DISPLAY MEMORY USED  
  Serial.printf("\n\n>> HEAP USED: %d\n", heapStart - ESP.getFreeHeap());  
  Serial.printf(">> PSRAM USED: %d", psramStart - ESP.getFreePsram());
  

  // CASE 1: Push sprite1 then sprite2 with transparency >> WORKS OK!
  sp1.pushSprite(60,20);
  sp2.pushSprite(centerX-50, centerY-50, TFT_TRANSPARENT);

  delay(2000); tft.fillScreen(TFT_BLACK);

  // CASE 2: Push sprite1 then rotated sprite2 with transparency >> TRANSP IS LOST  
  sp1.pushSprite(60,20);
  sp2.pushRotated(30, TFT_TRANSPARENT);

  delay(2000); tft.fillScreen(TFT_BLACK);

  // CASE 3: Push rotated sprite2 to sprite1 then push sprite1 to tft >> TRANSP IS LOST
  sp2.pushRotated(&sp1, 45, TFT_TRANSPARENT);  
  sp1.pushSprite(60,20);

}

void loop() {}

```

Program Output:

```
>>>> ESP-STARTED <<<<

> Internal Total heap 372096 kb, internal Free Heap 346040 
> SPIRam Total heap 4194252 kb, SPIRam Free Heap 4194252 
> ChipRevision 1, Cpu Freq 240, SDK Version v3.2.3-14-gd3e562907
> Flash Size 16777216 kb, Flash Speed 80000000 Hz
*** BOARD HAS PSRAM ***

>> HEAP USED: 528
>> PSRAM USED: 60040
```

Bodmer, 2020-09-14:
OK, I just ran the test and see the problem. Updates Sprite.cpp, it should now be fixed now.

Thanks for the example shjowing the problem, made it much easier to find the bug.

devbender, 2020-09-15:
No problem! Thanks for the quick fix.

flykarlos, 2021-03-01:
Hi Bodmer
I am using GC9A01 display. I realized that when I create sprite, draw rect symbolising needle with specific color and push it to the screen color is displayed as it should be. When I push same sprite, but rotated the color is different. Looks that positions of RGB components are swapped. Have a look on screenshot:
[image]

 Testing program included below.
```

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI(); 
TFT_eSprite neddle_SHORT = TFT_eSprite(&tft);

void setup() {

tft.begin();
tft.setRotation(2);
tft.setSwapBytes(true);
tft.setPivot(tft.width() / 2, tft.height() / 2);
tft.fillScreen(TFT_BLACK);

neddle_SHORT.createSprite(30, 130);
neddle_SHORT.setPivot(15, 100);
neddle_SHORT.fillRect(12, 0, 18, 130, TFT_GREEN);
neddle_SHORT.pushSprite(105,0);
delay(2000);
}

void loop() {

  for (int16_t i = 0; i <= 360; i += 5) {
      neddle_SHORT.pushRotated(i, TFT_WHITE);
      delay(5);
  }
}
```

What I am doing wrong?
thanks in advance!

flykarlos, 2021-03-01:
Ok, I think that command` tft.setSwapBytes(true);` is source of the problem. When commented out everything is ok. So it is needed to draw JPGs, and should be set to false after displaying JPEgs.

JasXSL, 2021-03-12:
This bug is back again.  The fix in commit 0c49b71dd434b938628377f655acb5186e4c0ddc (making tpcolor a uint16_t) was reverted in commit 63dc2120617717e760aa7e581710abbad2dced78

Bodmer, 2021-03-13:
Unfortunately the fix broke another condition, see #1022 hence the reversion, I need to find a solution that satisfies both scenarios!

commit 0c49b71d (2020-09-14): Fix #751