Bodmer/TFT_eSPI #752: undefined reference to `TFT_eSPI_Button::press(bool)'

sbonaime, 2020-09-16:
Hi,
I think I have a small problem with some TFT_eSPI exmaples. When I try compile TFT_Button_Label_Datum, I have the following error. 
````
sketch/TFT_Button_Label_Datum.ino.cpp.o:(.literal._Z4loopv+0x20): undefined reference to `TFT_eSPI_Button::press(bool)'
sketch/TFT_Button_Label_Datum.ino.cpp.o:(.literal._Z4loopv+0x24): undefined reference to `TFT_eSPI_Button::justPressed()'
sketch/TFT_Button_Label_Datum.ino.cpp.o:(.literal._Z4loopv+0x28): undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
sketch/TFT_Button_Label_Datum.ino.cpp.o:(.literal._Z4loopv+0x2c): undefined reference to `TFT_eSPI_Button::justReleased()'
sketch/TFT_Button_Label_Datum.ino.cpp.o:(.literal._Z11drawButtonsv+0x8): undefined reference to `TFT_eSPI_Button::initButton(TFT_eSPI*, short, short, unsigned short, unsigned short, unsigned short, unsigned short, unsigned short, char*, unsigned char)'
sketch/TFT_Button_Label_Datum.ino.cpp.o:(.literal._Z11drawButtonsv+0xc): undefined reference to `TFT_eSPI_Button::setLabelDatum(short, short, unsigned char)'
sketch/TFT_Button_Label_Datum.ino.cpp.o:(.literal.startup._GLOBAL__sub_I_tft+0x4): undefined reference to `TFT_eSPI_Button::TFT_eSPI_Button()'
sketch/TFT_Button_Label_Datum.ino.cpp.o: In function `loop()':
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:83: undefined reference to `TFT_eSPI_Button::press(bool)'
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:89: undefined reference to `TFT_eSPI_Button::justPressed()'
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:91: undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:95: undefined reference to `TFT_eSPI_Button::justReleased()'
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:97: undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
sketch/TFT_Button_Label_Datum.ino.cpp.o: In function `drawButtons()':
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:117: undefined reference to `TFT_eSPI_Button::initButton(TFT_eSPI*, short, short, unsigned short, unsigned short, unsigned short, unsigned short, unsigned short, char*, unsigned char)'
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:121: undefined reference to `TFT_eSPI_Button::setLabelDatum(short, short, unsigned char)'
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:125: undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
sketch/TFT_Button_Label_Datum.ino.cpp.o: In function `_GLOBAL__sub_I_tft':
/var/folders/0v/9kwpf1l100v2cdwflp0y3d8w0000gn/T/arduino_modified_sketch_949692/TFT_Button_Label_Datum.ino:42: undefined reference to `TFT_eSPI_Button::TFT_eSPI_Button()'
collect2: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte ESP32 Wrover Module
````

I don't have those errors when running other examples like  like TFT_meters but the don't have buttons. 
I wrote my own sketch without problem until I used some TFT_eSPI_Button . Then I have this  errror
```` 
sketch/Filling.ino.cpp.o:(.literal._Z6menu_gPPc+0x28): undefined reference to `TFT_eSPI_Button::TFT_eSPI_Button()'
sketch/Filling.ino.cpp.o:(.literal._Z6menu_gPPc+0x2c): undefined reference to `TFT_eSPI_Button::initButton(TFT_eSPI*, short, short, unsigned short, unsigned short, unsigned short, unsigned short, unsigned short, char*, unsigned char)'
sketch/Filling.ino.cpp.o:(.literal._Z6menu_gPPc+0x30): undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
sketch/Filling.ino.cpp.o: In function `menu_g(char**)':
/Arduino/Filling/Filling.ino:399: undefined reference to `TFT_eSPI_Button::TFT_eSPI_Button()'
/Arduino/Filling/Filling.ino:399: undefined reference to `TFT_eSPI_Button::initButton(TFT_eSPI*, short, short, unsigned short, unsigned short, unsigned short, unsigned short, unsigned short, char*, unsigned char)'
Arduino/Filling/Filling.ino:399: undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
/Arduino/Filling/Filling.ino:399: undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
collect2: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte ESP32 Wrover Module````

Bodmer, 2020-09-16:
Are you using the Arduino IDE?

sbonaime, 2020-09-17:
@Bodmer yes

Bodmer, 2020-09-17:
The TFT_eSPI library does not use Adafruit_GFX so should not be included. The button class is part of TFT_eSPI.

sbonaime, 2020-09-18:
@Bodmer I have the error when I compile exactly the TFT_Button_Label_Datum  example. Any idea ? thank you

Bodmer, 2020-09-18:
I am unable to reproduce the error, the sketch compiles fine ( I am using IDE version1.8.13). I suspect there is a conflict somewhere in your setup, but am unable to suggest what it could be. I would create a clean install of TFT_eSPI, but make sure you save a copy of your configuration settings.

sbonaime, 2020-09-19:
OK I give up with this problem. It should be an installation problem. Thanks

closed 2020-09-19 by Bodmer

jack828, 2021-08-01:
Hi. I have found the source of this error. In TFT_eSPI.cpp:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L4972-L4975

The button definitions are only included if `TOUCH_CS` is defined.

Since I am not using the built in SPI touch support, that was not defined for me.
Simply move that `#include` outside the `#if` and you will have the `TFT_eSPI_Button` class.

Maybe this should be included by default if the button code is not dependant on the libraries built-in touch controllers?

Bodmer, 2022-01-08:
I think the original problem has been fixed now in v2.4.23 for some reason the class linkage got lost.

jack828, 2022-03-12:
Hey @Bodmer I've just come back to using your project (thank you again 🙏🏻) however the issue remains (TFT_eSPI@2.4.42), and I don't believe it to be a linking error. I think this is still best fixed by the above change - allowing users access to the button extensions even if `TOUCH_CS` support is not provided in your library/by the display driver.

I think it would be beneficial for a user toggle-able include like `TFT_BUTTONS` or something if the ROM usage increase is something that may concern you or users.

What are your thoughts?

Bodmer, 2022-03-13:
The reported errors for the TFT_Button_Label_Datum example  are a bit odd because it will not produce errors associated with the button class if TOUCH_CS is defined. Instead, if TOUCH_CS is not defined the compiler halts on errors with the touch functions, e.g.:
```
...TFT_Button_Label_Datum:76:22: error: 'class TFT_eSPI' has no member named 'getTouch';...
```
Since the OP did not get that error, then TOUCH_CS must have been defined in the setup file.  Thus it makes no sense that the button functions are undefined because by defining TOUCH_CS the button class gets automatically included!   I now guess the OP must had modified the TFT_Button_Label_Datum example sketch, perhaps using a third party touch library or had commented out the touch code functions, hence the new unexpected error list.

I think including the button class by default is, as you suggest, needed so that other touch libraries can be used with the button class. Tests show with an ESP32 it only adds 56 bytes to the program size and consumes zero RAM if it is not used. These days that is negligible, so I have made that change and the button class is always included.  If my guess is right I think that will also help the OP.