Bodmer/TFT_eSPI #754: Partially draw complex object on small sprite.

dr-gino, 2020-09-21:
I have a function that draws a fairly complex object (a graph), it's not very suited to 4bpp since it uses VLW fonts, I'm assuming the anti aliasing is not going to work nicely in a 4bpp plane or at least harder to manage because of the palette? To avoid flickering I wanted to create a sprite but there's not enough RAM available for the entire image.
Is it possible to create a small sprite and then somehow reposition the origin before drawing so everything drawn "outside" of the sprite just gets ignored? This way I could potentially run the function twice on a smaller sprite and push it to the screen twice, completing the final image on screen without any flickering.

Bodmer, 2020-09-21:
Yes, it is possible. Example below.

```
/*
  Sketch to show how a small Sprite can be used in "pages" to
  update a larger area of the screen.

  Example for library:
  https://github.com/Bodmer/TFT_eSPI
*/

// Set delay after plotting the sprite
#define DELAY 1000

// Width and height of sprite
#define WIDTH  240
#define HEIGHT 10

// Sprite rows to draw (24 * HEIGHT = 240), so updated area is 240 * 240
#define TOTAL_ROWS   24

#define ROW_DELAY 1000                // Slow down row draws so they can be seen

#include <TFT_eSPI.h>                 // Include the graphics library (this includes the sprite functions)

TFT_eSPI    tft = TFT_eSPI();         // Declare object "tft"

TFT_eSprite spr = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

void setup()
{
  Serial.begin(250000);
  Serial.println();

  // Initialise the TFT registers
  tft.init();

  // Optionally set colour depth to 8 or 16 bits, default is 16 if not spedified
  // spr.setColorDepth(8);

  // Create a sprite of defined size
  spr.createSprite(WIDTH, HEIGHT);

  // Clear the TFT screen to blue
  tft.fillScreen(TFT_BLUE);
}

void loop(void)
{
  // Clear the TFT screen to blue
  tft.fillScreen(TFT_RED);

  for (uint16_t row = 0; row < TOTAL_ROWS; row ++)
  {
    // Fill the whole sprite with black (Sprite is in memory so not visible yet)
    spr.fillSprite(TFT_BLACK);

    // Draw a circle in the Sprite, pixels outside of the sprite are ignored
    // The centre of the circle is moved up the sprite heightas the rows are
    // drawn down the screen.
    //                   x,                        y,      radius,    colour
    spr.drawCircle(WIDTH/2, WIDTH/2 - (HEIGHT * row), WIDTH/2 - 5, TFT_GREEN);

    // Push the sprite to the TFT screen
    spr.pushSprite(0, HEIGHT * row);

    delay(ROW_DELAY);
  }

  delay(2000);
}
```

dr-gino, 2020-09-22:
I got it to work but setting the text datum to anything other than TL_DATUM causes all text that is supposed to be outside the sprite to end up inside the sprite.
These [lines ](https://github.com/Bodmer/TFT_eSPI/blob/0c49b71dd434b938628377f655acb5186e4c0ddc/TFT_eSPI.cpp#L3772-L3776)in TFT_eSPI::drawString are only executed if the datum/padding  is anything other than the default:
```
// Check coordinates are OK, adjust if not
if (poX < 0) poX = 0;
if (poX+cwidth > width())   poX = width() - cwidth;
if (poY < 0) poY = 0;
if (poY+cheight-baseline> height()) poY = height() - cheight;
```
So any text drawn with a datum other than TL_DATUM will be drawn inside the visible area of the sprite. Which is exactly what we don't want when partially drawing something.
The lines confining the coordinates to positive numbers only seem to be part of your initial commit and the other two were part of fix #122.

Bodmer, 2020-09-22:
As a temporary fix comment out those lines ([here for clarity of location](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L3772-L3776)) and you should be fine.

I will think further on this issue, the lines are probably not needed any more since all graphics should now be clipped to the TFT or sprite size.

Bodmer, 2020-09-22:
Test sketch indicates this works:
[Paged_Sprite.zip](https://github.com/Bodmer/TFT_eSPI/files/5261397/Paged_Sprite.zip)

Bodmer, 2020-10-02:
Updated library.

closed 2020-10-02 by Bodmer