Bodmer/TFT_eSPI #756: Newbie SPI TFT help request

sensboston, 2020-09-24:
Hi guys, I'm very sorry to open new issue (it's not about library but my lack of experience) but I really need some help, to get ESP32 dev board and ST7789 SPI TFT display working.

I'm trying connect [ESP32 devkit board v1](https://www.amazon.com/MELIFE-Development-Dual-Mode-Microcontroller-Integrated/dp/B07Q576VWZ) to [240x240 ST7789 SPI TFT display](https://www.amazon.com/gp/product/B088CQ4GPT/) but got stuck: display remains black, and I have no output at all :(

Please note (check Amazon links above, there are large pictures of ESP32 and TFT I'm using), this display doesn't have TFT_CS pin. So, on my **User_Setup.h** I defined:
```
#define TFT_CS   -1  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
```
and TFT DC pin connected to GPIO_02, TFT RES pin - to GPIO_04 (it looks like devkit v1 don't have a RST pin). VCC and BKL connected to ESP32 3V3, and GND to GND. 
But I **have no idea** how to connect TFTs SDA & SCK pins, and can't find any references with explanation how these pins should be used. I've googled for tutorials but lot of 'em have incorrect pin connection pictures.

P.S. [Actual picture of my setup](https://photos.app.goo.gl/sKo8P9iwnwEdj1MG7)

sukesh-ak, 2020-09-24:
Looks like your display is already configured to work with the library  
[Check here](https://github.com/Bodmer/TFT_eSPI/blob/0c49b71dd434b938628377f655acb5186e4c0ddc/User_Setup_Select.h#L52)

sensboston, 2020-09-24:
Thanks for advice, @sukesh-ak but still no luck at all. According to **Setup24_ST7789.h** (I left only line you've mentioned uncommented), I connected only 4 pins: VSS, GND, TFT_DC to GPIO5 (i.e. PIN_D1 - this Arduino-based inconsistency is killing me all the time!), and TFT_RES to GPIO2 (i.e. PIN_D4), rebuilded and uploaded sketch - still nothing :(
Two TFT pins aren't used at all, SCK & SDA.

So, my issue isn't resolved. Maybe someone can provide more detailed advice? 

P.S. I'm 100% sure what my ESP32 is fully functional and working, but don't know how to test TFT display (I've bought it from Amazon's seller in my first post).

sukesh-ak, 2020-09-24:
You can use this setup I believe.
So [Go here](https://github.com/Bodmer/TFT_eSPI/blob/0c49b71dd434b938628377f655acb5186e4c0ddc/User_Setup_Select.h#L52) and uncomment [this line ](https://github.com/Bodmer/TFT_eSPI/blob/0c49b71dd434b938628377f655acb5186e4c0ddc/User_Setup_Select.h#L52)

Generic setup is also given here. 
https://github.com/Bodmer/TFT_eSPI/blob/0c49b71dd434b938628377f655acb5186e4c0ddc/User_Setups/Setup24_ST7789.h#L20

Read the explanation in the header files. It should help. Unfortunately I don't have that display to try.

sensboston, 2020-09-24:
I tried to, but can't make things work (I believe, I explained that clearly in my comment above). In the **User_Setup_Select.h** I commented line 22:
`//#include <User_Setup.h>           // Default setup is root library folder`
and uncommented line 52:
`#include <User_Setups/Setup24_ST7789.h>            // Setup file configured for ST7789 240 x 240`

I also connected pins (described above, have only 4 pins connected: GND, VCC, RES & DC), opened Colour_Test example, compiled and loaded to ESP32 - nothing happened.

I'm asking here not because I'm stupid or can't read, or google for the samples, but because I just got stuck :(
It looks like I'm missing some important information...

sensboston, 2020-09-25:
@sukesh-ak, at [this manual](https://simple-circuit.com/esp8266-nodemcu-st7789-tft-ips-display/) I found sentence "The other display pins (SDA and SCL) are connected to NodeMCU hardware SPI pins (respectively D7 and D5)"
What you can tell about these pins (SDA & SCL, but it named SCK on my TFT)? I tried to connect 'em to GPIO13 & GPIO14 (should be ESP32 devkit hardware HSPI_CLK & HSPI_MOSI) but still no luck at all :(

sukesh-ak, 2020-09-25:
See if this helps.  
Sounds similar to yours since its SPI display with pinouts marked as I2C

https://youtu.be/HoZhgNcJjNA

sensboston, 2020-09-25:
@sukesh-ak, great find! Thanks, man, this is exactly what I want - got everything working. And this video is pretty good and clear; might be, @Bodmer can add it to wiki (especially for cases like mine).

sukesh-ak, 2020-09-25:
> @sukesh-ak, great find! Thanks, man, this is exactly what I want - got everything working. And this video is pretty good and clear; might be, @Bodmer can add it to wiki (especially for cases like mine).

Glad you got it working. The display board is a knock off by someone in a hurry, thus the pinout label is wrong.

closed 2020-09-27 by Bodmer