Bodmer/TFT_eSPI #759: int32_t & uint32_t compare in Sprite Class

Chipteck-CA, 2020-09-27:
With PlatformIO

TFT_eSPI\Extensions/Sprite.cpp:849:15: warning: comparison of integer expressions of different signedness: 'int32_t' {aka 'long int'} and 'uint32_t' {aka 'long unsigned int'} [-Wsign-compare]
  849 |       if ((sw > ds) && (_xe&1)) de = 1;

Chipteck-CA, 2020-09-28:
I think that I have found the problem and made changes as below,
 but would like someone to confirm this since I can build it and it is still incomplete with the following error:
scons: *** [.pio\build\bluepill_f103c8_128k\firmware.elf] 
	Implicit dependency `C:\users\frank_000.000\_.platformio\platforms\ststm32\builder\12000000' not found,_
	needed by target `.pio\build\bluepill_f103c8_128k\firmware.elf'.

In Sprite.cpp local variable declerations:

bool TFT_eSprite::pushSprite(int32_t tx, int32_t ty, int32_t sx, int32_t sy, int32_t sw, int32_t sh)
      //uint32_t ds = _xs&1; // Odd x start pixel
      int32_t ds = _xs&1; // Odd x start pixel
      //uint32_t de = 0;     // Odd x end pixel
      int32_t de = 0;     // Odd x end pixel
      //uint32_t dm = 0;     // Midsection pixel count
      int32_t dm = 0;     // Midsection pixel count
      //uint32_t yp = (_xs + ds + _iwidth * _ys)>>1;
      int32_t yp = (_xs + ds + _iwidth * _ys)>>1;
        //for (uint32_t dx = _xs; dx < _xs + sw; dx++) _tft->pushColor(readPixel(dx, _ys));
        for (int32_t dx = _xs; dx < _xs + sw; dx++) _tft->pushColor(readPixel(dx, _ys));

Bodmer, 2020-09-29:
The warning should not cause a problem as widths are always positive. I will add to my list of minor tweaks. Due to warning suppression they do not get flagged in the Arduino environment.

closed 2020-09-29 by Bodmer