Bodmer/TFT_eSPI #760: drawNumber - set number of digits

baniaque, 2020-09-28:
Thank you for awesome library - run it with blackpill f411 with one port use for parallel mode... and it's soooo fast and soooo furious :)
But i need display number with 3 digits, like 1 displayed as 001, 29 as 029 etc. 
Is there any function to do this? 
Something like drawNumber with additional parameter.

devbender, 2020-09-28:
For displaying leading zeros use the native C function sprintf to format a number as a string then simply print the string.

See bellow example.
```
int value = 29;
char buff[4];   // 3 digits total +1 null
sprintf(buff, "%03d, value); 
tft.print(buff);
```

Output should be: 029 for 2 digits or 009 for 1 digit.

baniaque, 2020-09-28:
@devbender I know that this can be done in some another ways but maybe this library got own solution for this, like LCDWiki. This is why I asking.

Bodmer, 2020-09-28:
Thanks for your positive feedback.

There is no function built into the TFT_eSPI library to do this, so it up to the user sketch to create a string in the format desired. There are many standard C functions to help.

To use the TFT_eSPI datum and padding functions you can use the code suggested above by @devbender and print to the TFT using drawString.

closed 2020-09-28 by Bodmer