Bodmer/TFT_eSPI #762: IL9143 & Touch & SD wont run

rs3hc, 2020-09-29:
**Hi Bodmer.
I tried to test the ESP32_SDcard_jpeg example bit wont work for me 
Example is untouched
SD CS pin is connected to GPIO 5

MISO, MOSI ans SCK Pins are shared to TFT and TOUCH pins

here my error:**
`
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
Card Mount Failed
ERROR: File "/EagleEye.jpg" not found!
ERROR: File "/Baboon40.jpg" not found!
ERROR: File "/lena20k.jpg" not found!
ERROR: File "/Mouse480.jpg" not found!

`

**my User_Setup.h**

`
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Define STM32 to invoke optimised processor support (only for STM32)
//#define STM32

// Defining the STM32 board allows the library to optimise the performance
// for UNO compatible "MCUfriend" style shields
//#define NUCLEO_64_TFT
//#define NUCLEO_144_TFT

// STM32 8 bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8 bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
//#define STM_PORTA_DATA_BUS
//#define STM_PORTA_DATA_BUS

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT

// Display type -  only define if RPi display
//#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
[... 260 lines cut ...]

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

`

Bodmer, 2020-09-29:
Run the SD library examples first to get the SD card interface working. Important: in the SD card library example setup() drive the TFT CS and Touch CS lines high at the start of setup() to avoid SPI bus conflict.

rs3hc, 2020-09-30:
Thanks Bodmer for helping

what example do you mean exactly?
i tried also Examples/SD(esp32)/SD_Test.ino... also didnt work.

I also closed JP1 on my TFT board to get 3,3v to SD Reader.... also didnt make it better..

here my circuit:
[image]

rs3hc, 2020-10-01:
Update:

I tried only ESP and SD with the SD.h included example (Arduino\hardware\espressif\esp32\libraries\SD\examples\SD_Test\SD_Test.ino).

so far it works!
my circuit:
[image]

Bodmer, 2020-10-02:
I suspect the SD library required use of the VSPI port (pins 5,18,19,23)

Now the SD card is working try adding the 
TFT, then the touch interface. Keep wires short to maintain signal integrity.

Bodmer, 2020-10-19:
Assumed resolved.

closed 2020-10-19 by Bodmer