Bodmer/TFT_eSPI #763: Can not run in stm32duino/L476RG,  st7789 135X240 always have half display?

KiraVerSace, 2020-09-29:
After five hours study, I still can not slove the problem.
I use the screen without rst, ST7789 1.14 and 135* 240 
Here is my setup:
```
#define STM32

#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 135

#define CGRAM_OFFSET      // Library will add offsets required

#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue

#define TFT_INVERSION_ON

#define TFT_SPI_PORT 2
#define TFT_MISO PB14
#define TFT_MOSI PB15
#define TFT_SCLK PB13
#define TFT_CS   PB12
#define TFT_DC   PC7
#define TFT_RST  -1  // Connect reset to ensure display initialises

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
// #define TFT_CS   -1      // Define as not used
// #define TFT_DC   PIN_D1  // Data Command control pin
// //#define TFT_RST  PIN_D4  // TFT reset pin (could connect to NodeMCU RST, see next line)
// #define TFT_RST  -1      // TFT reset pin connect to NodeMCU RST, must also then add 10K pull down to TFT SCK

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  		27000000

#define SPI_READ_FREQUENCY  20000000
//#define SPI_TOUCH_FREQUENCY  2500000

#define SUPPORT_TRANSACTIONS
```

and here is my main.c

```
TFT_eSPI tft = TFT_eSPI();

void setup(void)
{
	systemInitX();
	//socketLTE.socketInit();
	//socketLTE.readProperty(&lteProperty);

	sht20.begin(sht20.RESOLUTION_12BITS, SHT20_I2C, sht20Wire);

	tft.begin();
	tft.setRotation(3);

	tft.fillScreen(TFT_GREEN);
}
```

thisi is display picture
https://imgchr.com/i/0eyfMQ

KiraVerSace, 2020-09-29:
after try too many times,include changing the row and column offset, I think it's a bug.

KiraVerSace, 2020-09-29:
If I change the width and height to 360 360 , it can fill all the screen.

Please give me some tips.

Bodmer, 2020-09-29:
Define the display size in prtrait orientation as noted [here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/SetupX_Template.h#L70) and in example https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup135_ST7789.h#L5-L6.

The display driver addresses a total area of 240 x 320 pixels and different suppliers map the actual display pixels to different positions in the total area. So if the change above does not work then you have a new display variant and the offsets for your display will need to be determined empirically (to do this set the size to 240x 320 and determine the coordinate of the top left pixel in the display by creating a test sketch..

KiraVerSace, 2020-09-30:
Follow your tips, I change the width to 135 and the height 240, the problem is still exist, and if I comment the **#define CGRAM_OFFSET**, the display will be abnormally,the picture is uncomment.

Please help me,if I use the library adafruit-st7789, it can work!

Thank you for your help.[image]

Bodmer, 2020-10-01:
Down load the latest (2.2.20 )library as a zip file direct from Github.
Use [this setup](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup135_ST7789.h) as the basis for your tests with NO modifications.
Run the "[Read_User_Setup](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup)" exmple to make sure you have selected the correct setup in "User_Setup_Select.h", post the serial monitor output here so I can see all the settings you are using.

closed 2020-10-02 by Bodmer

KiraVerSace, 2020-10-02:
yeap, thank you for you fix the bug, I will try to enable the DMA in stm32 to drive the screen, improve the fps!
https://www.bilibili.com/video/BV1fa4y1j72c
You can see this video.
Thank you  very much!

commit ef93dbe6 (2020-10-02): Make ST7789 drivers consistent

Simplified driver ST7789_2_DRIVER not supports 135 x 240 display.
This will not fix #763 since user specifies alternative ST7789_DRIVER which already supports 135x240 displays.