Bodmer/TFT_eSPI #768: Driver 7789 non centered img

pocketPlay, 2020-10-02:
hi there,
I wrote a small example in July that showed an img in the center of the display.
With the update to the 7789 drivers no longer see the centered image, it is shifted up by about 40 pixels

I loaded the old drivers and the image is centered.

hardware always the same and no changes to the code

I also loaded the examples and the problem is the same, the screen is not centered.

esp32
TFT 7789 - 240X240
Any idea? Bug?

All the best

Bodmer, 2020-10-02:
This may be related to #763 which TFT_eSPI version worked.

pocketPlay, 2020-10-02:
hi,
the version of library is: 2.2.14

Regards

Bodmer, 2020-10-02:
There is no change to the drivers that I can see between 2.2.14 and the latest 2.2.21.

Are you sure that it is version 2.2.14 that worked for you?

Bodmer, 2020-10-02:
Note you must [include the display size](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup24_ST7789.h#L5-L6) in your setup file to invoke the offsets.

pocketPlay, 2020-10-05:
[image]
[image]

Sorry for the wait. Yes the old driver works for me.
I just overwrote the old driver folder on the new one (I always make a backup) and everything is back to working.

Here correct img with old driver 2.2.14 and new driver (img shifted)

All the best

[image]
[image]

Bodmer, 2020-10-19:
Without an example sketch that demonstrates the problem I am unable to advise.

closed 2020-10-25 by Bodmer