Bodmer/TFT_eSPI #772: I2S LCD Mode

dracir9, 2020-10-03:
This isn't an actual issue but a possible improvement/feature request. I've recently gone through the esp32 technical reference manual and noticed that I2S features an [LCD Mode](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#b8) which may help in parallel data transmission. 
If I understood correctly you just have to configure the GPIO matrix to set some pins as data output. Then just fill a FIFO queue with the data and start the transmission. There would be no need to set and clear IO pins manually.

Have you ever tried this mode? Would there be any improvement over direct IO manipulation? Is this going to be added in the future?

Bodmer, 2020-10-04:
Yes, I have tried this mode and the results were disappointing. For a single pixel screen update the I2S method was 10x slower, this is because setting up the transaction table and initiating the transfer takes so many CPU cycles. Once the update area increases to anout 50x50 pixels (I.e. 5kbyte block to be transferred) then the performance is the same as bit bashing. For even larger blocks there is a small performance gain and foreground tasks continue how ever foreground performance appears degraded presumably because there are interrupts running to keep the small I2S buffer serviced.

I tried switching between I2Smode alfor large blocks and bit bashing for general purpose graphics but for each switch the IO matrix has to be reconfigured and the Espressif software seems to have a bug and subsequently fails to restart the I2S. The documentation of the software for I2S mode is also very poor. Reading from the TFT with I2S also failed, so in the end I abandoned I2S and focused on adding SPI DMA.

dracir9, 2020-10-04:
I see.
I2S features DMA capabilities too, would there be any improvement?
I first came to I2S LCD Mode while reading the datasheet for the newer ESP32-S2. It includes a newer processor, more pins and an improved DMA system. Are there any plans for this?

Bodmer, 2020-10-04:
The above comments were for a DMA based parallel I2S interface to the TFT. Direct writes to the I2S interface buffer were also slower than bit bashing for some reason.

Yes, I will support the ESP32-S2 but at the moment I do not have one. Hopefully it will be possible to toggle the pins faster than the standard ESP32 which is limited to a 50ns toggle time due to the 20MHz I/O clock. With more pins perhaps a 16 bit interface is also possible which would at least double the speed.

closed 2020-10-04 by Bodmer