Bodmer/TFT_eSPI #774: GC9A01_DRIVER

freemanzwin, 2020-10-05:
hi. 
(my english is google translator)

I tested the GC9A01 240*240 circle lcd.
It works fine in the environment below.
I hope the source is added and tested.

original source : GC9A01.h
reference source: Arduino_GFX

my platformio.ini 
--------------------------
```
framework = arduino
platform = espressif32
build_flags = 
	-DUSER_SETUP_LOADED
	-DGC9A01_DRIVER
	-DTFT_WIDTH=240
	-DTFT_HEIGHT=240

	-DSPI_FREQUENCY=80000000

	-DTFT_MISO=-1
	-DTFT_MOSI=13
	-DTFT_SCLK=14
	-DTFT_CS=15
	-DTFT_DC=21
	-DTFT_RST=22
	-DTFT_BL=5

	-DTFT_BL_PWM=-1
	-DTFT_BACKLIGHT_ON=HIGH
	-DLOAD_GLCD
	-DLOAD_GFXFF

```
=================================
TFT_eSPI.cpp
---------------
```
...

#elif defined (SSD1963_800ALT_DRIVER)
    #include "TFT_Drivers/SSD1963_Init.h"

_**#elif defined (GC9A01_DRIVER)
     #include "TFT_Drivers/GC9A01_Init.h"**_

#endif

#ifdef TFT_INVERSION_ON
  writecommand(TFT_INVON);
#endif

...

#elif defined (ST7789_2_DRIVER)
    #include "TFT_Drivers/ST7789_2_Rotation.h"

**_#elif defined (GC9A01_DRIVER)
     #include "TFT_Drivers/GC9A01_Rotation.h"_**

#endif

  delayMicroseconds(10);

...
```
====================================
User_Setup_Select.h
------------------------
```
...
#elif defined (SSD1963_800ALT_DRIVER)
     #include "TFT_Drivers/SSD1963_Defines.h"
     #define  TFT_DRIVER 0x1963
**_#elif defined (GC9A01_DRIVER)
     #include "TFT_Drivers/GC9A01_Defines.h"
     #define  TFT_DRIVER 0x9A01_**
                              // <<<<<<<<<<<<<<<<<<<<<<<< ADD NEW DRIVER HERE
                              // XYZZY_init.h and XYZZY_rotation.h must also be added in TFT_eSPI.cpp
#elif defined (XYZZY_DRIVER)
     #include "TFT_Drivers/XYZZY_Defines.h"
     #define  TFT_DRIVER 0x0000
#else
...
```

====================================
Setup46_GC9A01.h
---------------------
```
// GC9A01 240 x 240 display

#define GC9A01_DRIVER     // Configure all registers

//#define TFT_WIDTH  240
//#define TFT_HEIGHT 240

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// #define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

// #define SUPPORT_TRANSACTIONS

```
==============================
GC9A01_Rotation.h
--------------------
```
  rotation = m % 4;

  writecommand(TFT_MADCTL);
  switch (rotation) {
    case 0: // Portrait
      writedata(TFT_MAD_BGR);
      _width  = _init_width;
      _height = _init_height;
      break;
    case 1: // Landscape (Portrait + 90)
      writedata(TFT_MAD_MX | TFT_MAD_MV | TFT_MAD_BGR);
      _width  = _init_height;
      _height = _init_width;
      break;
    case 2: // Inverter portrait
      writedata(TFT_MAD_MX | TFT_MAD_MY | TFT_MAD_BGR);
      _width  = _init_width;
      _height = _init_height;
      break;
    case 3: // Inverted landscape
      writedata(TFT_MAD_MV | TFT_MAD_MY | TFT_MAD_BGR);
      _width  = _init_height;
      _height = _init_width;
      break;
  }
```

==========================================
GC9A01_Init.h
---------------
```
{
  writecommand(0xEF);
  writecommand(0xEB);
  writedata(0x14);

  writecommand(0xFE);
  writecommand(0xEF);

  writecommand(0xEB);
  writedata(0x14);

  writecommand(0x84);
  writedata(0x40);

  writecommand(0x85);
  writedata(0xFF);

  writecommand(0x86);
  writedata(0xFF);

  writecommand(0x87);
  writedata(0xFF);

  writecommand(0x88);
  writedata(0x0A);

  writecommand(0x89);
  writedata(0x21);

  writecommand(0x8A);
  writedata(0x00);

  writecommand(0x8B);
  writedata(0x80);

  writecommand(0x8C);
  writedata(0x01);

  writecommand(0x8D);
  writedata(0x01);

  writecommand(0x8E);
  writedata(0xFF);

  writecommand(0x8F);
  writedata(0xFF);

  writecommand(0xB6);
  writedata(0x00);
  writedata(0x20);

  writecommand(0x3A);
  writedata(0x05);

  writecommand(0x90);
  writedata(0x08);
  writedata(0x08);
  writedata(0x08);
  writedata(0x08);

  writecommand(0xBD);
  writedata(0x06);

  writecommand(0xBC);
  writedata(0x00);

  writecommand(0xFF);
  writedata(0x60);
  writedata(0x01);
  writedata(0x04);

  writecommand(0xC3);
  writedata(0x13);
  writecommand(0xC4);
  writedata(0x13);

  writecommand(0xC9);
  writedata(0x22);

  writecommand(0xBE);
  writedata(0x11);

  writecommand(0xE1);
  writedata(0x10);
  writedata(0x0E);

  writecommand(0xDF);
  writedata(0x21);
  writedata(0x0c);
  writedata(0x02);

  writecommand(0xF0);
  writedata(0x45);
  writedata(0x09);
  writedata(0x08);
  writedata(0x08);
  writedata(0x26);
  writedata(0x2A);

  writecommand(0xF1);
  writedata(0x43);
  writedata(0x70);
  writedata(0x72);
  writedata(0x36);
  writedata(0x37);
  writedata(0x6F);

  writecommand(0xF2);
  writedata(0x45);
  writedata(0x09);
  writedata(0x08);
  writedata(0x08);
  writedata(0x26);
  writedata(0x2A);

  writecommand(0xF3);
  writedata(0x43);
  writedata(0x70);
  writedata(0x72);
  writedata(0x36);
  writedata(0x37);
  writedata(0x6F);

  writecommand(0xED);
  writedata(0x1B);
  writedata(0x0B);

  writecommand(0xAE);
  writedata(0x77);

  writecommand(0xCD);
  writedata(0x63);

  writecommand(0x70);
  writedata(0x07);
  writedata(0x07);
  writedata(0x04);
  writedata(0x0E);
  writedata(0x0F);
  writedata(0x09);
  writedata(0x07);
  writedata(0x08);
  writedata(0x03);

  writecommand(0xE8);
  writedata(0x34);

  writecommand(0x62);
  writedata(0x18);
  writedata(0x0D);
  writedata(0x71);
  writedata(0xED);
  writedata(0x70);
  writedata(0x70);
  writedata(0x18);
  writedata(0x0F);
  writedata(0x71);
  writedata(0xEF);
  writedata(0x70);
  writedata(0x70);

  writecommand(0x63);
  writedata(0x18);
  writedata(0x11);
  writedata(0x71);
  writedata(0xF1);
  writedata(0x70);
  writedata(0x70);
  writedata(0x18);
  writedata(0x13);
  writedata(0x71);
  writedata(0xF3);
  writedata(0x70);
  writedata(0x70);

  writecommand(0x64);
  writedata(0x28);
  writedata(0x29);
  writedata(0xF1);
  writedata(0x01);
  writedata(0xF1);
  writedata(0x00);
  writedata(0x07);

  writecommand(0x66);
  writedata(0x3C);
  writedata(0x00);
  writedata(0xCD);
  writedata(0x67);
  writedata(0x45);
  writedata(0x45);
  writedata(0x10);
  writedata(0x00);
  writedata(0x00);
  writedata(0x00);

  writecommand(0x67);
  writedata(0x00);
  writedata(0x3C);
  writedata(0x00);
  writedata(0x00);
  writedata(0x00);
  writedata(0x01);
  writedata(0x54);
  writedata(0x10);
  writedata(0x32);
  writedata(0x98);

  writecommand(0x74);
  writedata(0x10);
  writedata(0x85);
  writedata(0x80);
  writedata(0x00);
  writedata(0x00);
  writedata(0x4E);
  writedata(0x00);

  writecommand(0x98);
  writedata(0x3e);
  writedata(0x07);

  writecommand(0x35);
  writecommand(0x21);

  writecommand(0x11);
  delay(120);
  writecommand(0x29);
  delay(20);
}

```
===========================
GC9A01_Defines.h
----------------------
```
// Change the width and height if required (defined in portrait mode)
// or use the constructor to over-ride defaults
#ifndef TFT_WIDTH
  #define TFT_WIDTH  240
#endif
#ifndef TFT_HEIGHT
  #define TFT_HEIGHT 320
#endif

// Delay between some initialisation commands
#define TFT_INIT_DELAY 0x80

// Generic commands used by TFT_eSPI.cpp
#define TFT_NOP     0x00
#define TFT_SWRST   0x01

#define TFT_CASET   0x2A
#define TFT_PASET   0x2B
#define TFT_RAMWR   0x2C

#define TFT_RAMRD   0x2E
#define TFT_IDXRD   0x00 //0xDD // ILI9341 only, indexed control register read

#define TFT_MADCTL  0x36
#define TFT_MAD_MY  0x80
#define TFT_MAD_MX  0x40
#define TFT_MAD_MV  0x20
#define TFT_MAD_ML  0x10
#define TFT_MAD_BGR 0x08
#define TFT_MAD_MH  0x04
#define TFT_MAD_RGB 0x00

#define TFT_INVOFF  0x20
#define TFT_INVON   0x21

```

Bodmer, 2020-10-05:
Thanks, I will add into a future release. I see there are some small round displays with that controller fitted.

Bodmer, 2020-10-25:
Added.

Can you test please as I do not have one of those displays.

Bodmer, 2020-10-31:
I assume it is working.

closed 2020-10-31 by Bodmer

poky, 2020-12-17:
Hi, thanks for the great library & driver!
I have some issues running the drawXBitmap example, it only shows the blue lines jumping in different locations, was wondering how to debug, any thoughts will be appreciated!

[image]

Bodmer, 2020-12-17:
Run the "Read_User_Setup" diagnostic sketch and check the compiler is picking up you setup file correctly.

It looks like you have not soldered the header to the board, this will give reliability issues.

Keep the wires to the board short (less that 100mm) to avoid signal inegrity issues.

poky, 2020-12-17:
> Run the "Read_User_Setup" diagnostic sketch and check the compiler is picking up you setup file correctly.
> 
> It looks like you have not soldered the header to the board, this will give reliability issues.
> 
> Keep the wires to the board short (less that 100mm) to avoid signal inegrity issues.

Hi, here's the report:
[code]
TFT_eSPI ver = 2.3.51
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9A01
Display width  = 240
Display height = 240

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]

Bodmer, 2020-12-17:
The setup looks OK. Other than trying a lower SPI frequency I have no ideas what could be wrong.

I do not have one of those displays, maybe @freemanzwin will be able to help.

freemanzwin, 2021-03-15:
This Chinese LCD module does not have MISO pin. So I don't think I can read CGRAM.