Bodmer/TFT_eSPI #775: 9341 Don´t work

rasuncion19, 2020-10-05:
My TFT-ILI9341 Don´t work with this library. it just goes black, I am working on esp32. 

VCC= 3.3V
LED=3.3V
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC   4  // Data Command control pin
#define TFT_RST   2  // Reset pin (could connect to RST pin)

what am I doing wrong?
The examples themselves do nothing

rasuncion19, 2020-10-05:
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER

// Some displays support SPI reads via the MISO pin, other displays have a single
// bi-directional SDA pin and the library will try to read this via the MOSI line.
// To use the SDA line for reading data from the TFT uncomment the following line:

// #define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 display only

// For ST7789 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display

//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

// For M5Stack ESP32 module with integrated ILI9341 display ONLY, remove // in line below

// #define M5STACK

// For ST7789, ST7735 and ILI9163 ONLY, define the pixel width and height in portrait orientation
// #define TFT_WIDTH  80
// #define TFT_WIDTH  128
// #define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
// #define TFT_HEIGHT 160
// #define TFT_HEIGHT 128
// #define TFT_HEIGHT 240 // ST7789 240 x 240
// #define TFT_HEIGHT 320 // ST7789 240 x 320

// For ST7735 ONLY, define the type of display, originally this was based on the
// colour of the tab on the screen protector film but this is not always true, so try
// out the different options below if the screen does not display graphics correctly,
// e.g. colours wrong, mirror images, or tray pixels at the edges.
// Comment out ALL BUT ONE of these options for a ST7735 display driver, save this
// this User_Setup file, then rebuild and upload the sketch to the board again:

// #define ST7735_INITB
// #define ST7735_GREENTAB
// #define ST7735_GREENTAB2
// #define ST7735_GREENTAB3
// #define ST7735_GREENTAB128    // For 128 x 128 display
// #define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 offset)
// #define ST7735_REDTAB
// #define ST7735_BLACKTAB
[... 179 lines cut ...]
 #define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

Bodmer, 2020-10-05:
Run the diagnostic sketch Read_User_Setup which outputs diagnostic information to the serial monitor window. Check all pins are listed as you expect.  Post output here.

rasuncion19, 2020-10-05:
[code]
TFT_eSPI ver = 2.2.14
Processor    = ESP50
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D9 (GPIO 23)
MISO    = D9 (GPIO 19)
SCK     = D9 (GPIO 18)
TFT_CS   = D8 (GPIO 15)
TFT_DC   = D3 (GPIO 0)
TFT_RST  = D4 (GPIO 2)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
[/code]

Bodmer, 2020-10-05:
The diagnostic sketch indicates the software is expect the display DC pin to be wired to GPIO 0. Check the right setup file is being used or rewire DC to GPIO 0.

rasuncion19, 2020-10-05:
I've rewired and it still doesn't work

Bodmer, 2020-10-05:
Check the User_Setup_Select.h file has been editted correctly to calling up ***ONE*** setup file. I think it is pointing to the default User_Setup.h file which is the settings for an ESP8266.

Provide more inforrmation on your software setup:

- Arduino IDE version
- ESP32 board package version
- Pictures of your boards and wiring

rasuncion19, 2020-10-06:
Versión de Arduino IDE --> 1.8.13

[pantalla](https://ibb.co/pdFvCdt)
[Esp32](https://ibb.co/C9jXDxx)

Bodmer, 2020-10-06:
Does the tft backlight glow?

rasuncion19, 2020-10-06:
yes

Bodmer, 2020-10-06:
I can't see anything obvious wrong. Using GPIO 0 may be an issue as it has alternative functions at boot time. Edit Setup.h for the ESP32 using the GPIO i suggested, by default ESP8266 pins are defined and these must be commented out. Make sure ***only*** that setup header 
file is being called up in User_Setup_Select.h. Wire TFT using those pins. Run the Read_User_Setup again and check the pins are correct.

rasuncion19, 2020-10-06:
I have noticed, when I modify the set up, the read_user_setup don´t change

rasuncion19, 2020-10-06:
User_Setup_Select

// These are the pins for all ESP8266 boards
//      Name   GPIO    Function
#define PIN_D0  16  // WAKE
#define PIN_D1   5  // User purpose
#define PIN_D2   4  // User purpose
#define PIN_D3   0  // Low on boot means enter FLASH mode
#define PIN_D4   2  // TXD1 (must be high on boot to go to UART0 FLASH mode)
#define PIN_D5  14  // HSCLK
#define PIN_D6  12  // HMISO
#define PIN_D7  13  // HMOSI  RXD2
#define PIN_D8  15  // HCS    TXD0 (must be low on boot to enter UART0 FLASH mode)
#define PIN_D9   3  //        RXD0
#define PIN_D10  1  //        TXD0

#define PIN_MOSI 8  // SD1
#define PIN_MISO 7  // SD0
#define PIN_SCLK 6  // CLK
#define PIN_HWCS 0  // D3

#define PIN_D11  9  // SD2
#define PIN_D12 10  // SD4

rasuncion19, 2020-10-10:
Hello?

Bodmer, 2020-10-10:
I suspect you have not eddited the setup files correctly. As I cannot see them all I can only guess what you have done wrong.

Therefore to get back to a known state:
- Delete your old TFT_eSPI library, load a brand new copy of the library.
- ONLY edit the User_Setup.h file to suit your hardware and connections.
- Compile and upload sketch with this new setup

Bodmer, 2020-10-14:
Hello?

rasuncion19, 2020-10-14:
Thanks for your help, these days I have been busy, tomorrow I will try it without fail.

Bodmer, 2020-10-19:
Assumed to be a user error.

closed 2020-10-19 by Bodmer

Ramiiyan, 2022-10-30:
Hi All,
I also experienced the same. 
As i check we need to change the pins(Hardware) as below.
```
#define TFT_DC    0  // Data Command control pin
#define TFT_RST   2  // Reset pin (could connect to RST pin)
```
After that, it working as expected.