Bodmer/TFT_eSPI #776: I can't implementing the library

ahmetOzby, 2020-10-05:
I want to use stm32f103c series and I think I've done all the includes correctly. When I try to compile, I got this.

In file included from C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:25:0:
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:17:46: error: no matching function for call to 'SPIClass::SPIClass(<anonymous enum>, <anonymous enum>, <anonymous enum>)'
     SPIClass spi(TFT_MOSI, TFT_MISO, TFT_SCLK);
                                              ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:17:46: note: candidates are:
In file included from C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.h:28:0,
                 from C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:17:
C:\Users\Ahmet\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.8.9\libraries\SPI\src/SPI.h:176:5: note: SPIClass::SPIClass(uint32)
     SPIClass(uint32 spiPortNumber);
     ^
C:\Users\Ahmet\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.8.9\libraries\SPI\src/SPI.h:176:5: note:   candidate expects 1 argument, 3 provided
C:\Users\Ahmet\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.8.9\libraries\SPI\src/SPI.h:170:7: note: constexpr SPIClass::SPIClass(const SPIClass&)
 class SPIClass {
       ^
C:\Users\Ahmet\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.8.9\libraries\SPI\src/SPI.h:170:7: note:   candidate expects 1 argument, 3 provided
C:\Users\Ahmet\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.8.9\libraries\SPI\src/SPI.h:170:7: note: constexpr SPIClass::SPIClass(SPIClass&&)
C:\Users\Ahmet\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.8.9\libraries\SPI\src/SPI.h:170:7: note:   candidate expects 1 argument, 3 provided
In file included from C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:25:0:
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:20:3: error: 'SPI_HandleTypeDef' does not name a type
   SPI_HandleTypeDef spiHal;
   ^
In file included from C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:25:0:
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c: In member function 'void TFT_eSPI::pushBlock(uint16_t, uint32_t)':
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:349:23: error: 'spiHal' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
                       ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:349:54: error: 'HAL_MAX_DELAY' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
                                                      ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:349:67: error: 'HAL_SPI_Transmit' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
                                                                   ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:355:23: error: 'spiHal' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, BUF_SIZE<<1, HAL_MAX_DELAY);
                       ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:355:59: error: 'HAL_MAX_DELAY' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, BUF_SIZE<<1, HAL_MAX_DELAY);
                                                           ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:355:72: error: 'HAL_SPI_Transmit' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, BUF_SIZE<<1, HAL_MAX_DELAY);
                                                                        ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:359:30: error: 'spiHal' was not declared in this scope
   if (len) HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY); //*/
                              ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:359:61: error: 'HAL_MAX_DELAY' was not declared in this scope
   if (len) HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY); //*/
                                                             ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:359:74: error: 'HAL_SPI_Transmit' was not declared in this scope
   if (len) HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY); //*/
                                                                          ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c: In member function 'void TFT_eSPI::pushPixels(const void*, uint32_t)':
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:374:25: error: 'spiHal' was not declared in this scope
       HAL_SPI_Transmit(&spiHal, (uint8_t*)col, BUF_SIZE<<1, HAL_MAX_DELAY);
                         ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:374:61: error: 'HAL_MAX_DELAY' was not declared in this scope
       HAL_SPI_Transmit(&spiHal, (uint8_t*)col, BUF_SIZE<<1, HAL_MAX_DELAY);
                                                             ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:374:74: error: 'HAL_SPI_Transmit' was not declared in this scope
       HAL_SPI_Transmit(&spiHal, (uint8_t*)col, BUF_SIZE<<1, HAL_MAX_DELAY);
                                                                          ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:378:23: error: 'spiHal' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
                       ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:378:54: error: 'HAL_MAX_DELAY' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
                                                      ^
C:\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_STM32.c:378:67: error: 'HAL_SPI_Transmit' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
[... 160 lines cut ...]
 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, uint8_t transp, bool bpp8, uint16_t *cmap)
      ^
In file included from C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:4309:0:
C:\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp:837:76: error: call of overloaded 'pushImage(int32_t&, int32_t, int32_t&, int, uint8_t*, int)' is ambiguous
       _tft->pushImage(tx, ty++, sw, 1, _img8 + _xs + _iwidth * _ys++, true );
                                                                            ^
C:\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp:837:76: note: candidates are:
C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:1135:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, bool, uint16_t*)
 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, bool bpp8,  uint16_t *cmap)
      ^
C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:1292:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, uint8_t, bool, uint16_t*)
 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, uint8_t transp, bool bpp8, uint16_t *cmap)
      ^
In file included from C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:4309:0:
C:\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp:873:73: error: call of overloaded 'pushImage(int32_t&, int32_t&, int32_t&, int32_t&, uint8_t*, int)' is ambiguous
       _tft->pushImage(tx, ty, sw, sh, _img8 + (_iwidth>>3) * _ys, false );
                                                                         ^
C:\Arduino\libraries\TFT_eSPI-master\Extensions/Sprite.cpp:873:73: note: candidates are:
C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:1135:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, bool, uint16_t*)
 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, bool bpp8,  uint16_t *cmap)
      ^
C:\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:1292:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, uint8_t, bool, uint16_t*)
 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, uint8_t transp, bool bpp8, uint16_t *cmap)
      ^
exit status 1

Bodmer, 2020-10-05:
The library requires the official STM32 board package, not third party libraaries.

Install the STM32 board support from the Arduino IDE menu option. You should then have a package path like this:

```
C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SPI
                                               ^^^^^ Note!
```

ahmetOzby, 2020-10-06:
Thank you so much. You solved this immediately.

closed 2020-10-06 by ahmetOzby