Bodmer/TFT_eSPI #779: #define TFT_CS, TFT_RST, TFT_DC and TFT_RD as -1 ?

quangvankts, 2020-10-06:
Hi
Will a ILI9481 - ESP32 setup still work in 8 bit parallel mode if I connect CS pin to GND, DC and RD pin to Vcc, TFT_RST to EN and define them in User_Setup.h as -1 ?
Thanks

Bodmer, 2020-10-06:
No. As a minimum DC, WR and the data pins must be driven with CS pulled low, RST and RD pulled high.
CS, RST and RD can  then be defined  as -1

closed 2020-10-06 by Bodmer

quangvankts, 2020-10-06:
Thank you 
Is it possible to drive LCD’s DC pin through a shift register ?  If yes which part of the library can I modify to do so ?

Bodmer, 2020-10-06:
Yes, [these are the lines to adapt](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L81-L87), they are the macros that toggle the DC pin for an 8 bit parallel interface on an ESP32.

quangvankts, 2020-10-07:
Got it ! 
Thanks again