Bodmer/TFT_eSPI #783: Touch Screen for STM32

ahmetOzby, 2020-10-09:
Hi,

Can you say what is the touchscreen connection for STM32. There is no definition for stm32 touch pins in User_Setup.h.
How should I define the pins for running the Touch_Controller_Demo example?

Thank you for your time.

ahmetOzby, 2020-10-09:
Edit: The screen has the resistive touch pins. The pins are:
1) YU
2) XL
3) YD
4) XR

Bodmer, 2020-10-10:
It depends on the type of touch screen interface you have. The TFT_eSPI library only supports a SPI interface with the XPT2046 touch controller. Check your screen specifications or examine the back of the circuit board.

ahmetOzby, 2020-10-10:
Thank you for your answer Bodmer. 
I couldn't find the touch controller of HT0240BI01BR1. I want to clarify something.  The screen has YU, XL, YD and XR. Can I analog read these signals on stm32 or I have to do something else?

Bodmer, 2020-10-10:
OK, it looks like you only have access to the bare touch screen terminals. This is typical for UNO style shields. You need to look for touch support library for that interface type. Yes, thay are analogue interface pins but to extract the x and y positions pins have to be driven and read in a sequence.  Libraries such as Mcufriend_kbv support that interface type but I am not aware of a stand-alone library that handles that interface type.

closed 2020-10-10 by Bodmer