Bodmer/TFT_eSPI #784: ILI9341 (MRB3205) set to parallel 8bit - ESP32 image problem

KamaJogi, 2020-10-11:
Hi,

Arduino, libraries and ESP32 are up to date.
My ESP32 version -> https://components101.com/microcontrollers/esp32-devkitc
I use setup file -> Setup14_ILI9341_Parallel.h
LCD pins are soldered to the ESP32 on the board.
LCD VDD directly connected to 5V.
I had to set the ESP CPU Frequency to 40 MHz.
Higher CPU frequency = white screen on LCD.

[image]

Bodmer, 2020-10-11:
This is likely caused by one or more of the following:
1. Inadequate power supply to the ESP32 and TFT (not enough current)
2. Inadequate ground (0V or GND) connections between ESP32 and display (too long), add extra connections to avoid "ground bounce"
3. Wires from ESP32 to the TFT are too long causing "ringing" on signals
4. Unstable logic level on RD pin

KamaJogi, 2020-10-12:
1. Power 3.3V supply  3A  (Oscilloscope test - ok)
2. Inadequate ground (0V or GND) (Oscilloscope test - ok)
3. Wires from ESP32 -> long, short or soldered - no diffrence,
4. Stable logic level on RD pin (Oscilloscope test - ok).

I changed the pin assignments - the same symptoms :(

Maybe there is something wrong with ESP32 - but SPI displays works OK.

Maybe the logic state changes too fast for LCD - pins are directly connected to LCD BUS without HC245 -> How do I slow down the change of logic states in your code?

[image]

KamaJogi, 2020-10-12:
ESP32 - DevKitC was caused problem.

I found MH-ET LIVE D1 mini ESP32 - connect to LCD and everything works fine.

Thanks for help.

closed 2020-10-12 by KamaJogi