Bodmer/TFT_eSPI #785: Push a rotated sprite to a sprite?

DrJaymz, 2020-10-11:
I have used push rotated before but only when pushing a sprite to the main screen.

I now have a 200x200 square sprite called spr_horizon and I wanted to use another sprite (spr_text) to it.  Mainly just to create rotated text. I created the child sprite like this spr_text = TFT_eSprite(&spr_horizon);  For some reason I am getting it plotting the sprite outside of spr_horizon and the coordinates are a bit off, its like its drawing on the main TFT.

//this drawn non-rotated text is the right place.
    spr_horizon.drawString(string, x, y, 2);

    //fill my sprite
    spr_text.setTextDatum(MC_DATUM);
    spr_text.drawString(string, 0,0,2);
    
    //This draws outside the main sprite
    //spr_text.pushSprite(x,y);
    //this draws in another dimension or something.
    spr_text.setPivot(x,y);
    spr_text.pushRotated(-roll);

Am I doing this right?

The full block looks something like this:

//basically I'm plotting every 10 degrees on an artificial horizon.

 spr_text = TFT_eSprite(&spr_horizon);
  spr_text.setColorDepth(8);                            // Create an 8bpp Sprite
  spr_text.createSprite(20, 40); //

  for (int i = -10; i < 10; i++)
  {
    spr_text.fillScreen(TFT_MAGENTA);
    if(i >=0 ) 
    {
      spr_horizon.setTextColor(TFT_BLACK);
       sprintf(string, "-%d-", i* 10);
    }
    else
    {
       sprintf(string, "-%d-", -i* 10);
      spr_horizon.setTextColor(TFT_WHITE);
    }
       
    float th = -roll_radians - (PI / 2);
    int x = i * 20 * cos(th) + HORIZON_MID_X - 10;
    int y = i * 20 * sin(th) + HORIZON_MID_Y + pitchpixels- font_y_offset;
    
    //this drawn non-rotated text is the right place.
    spr_horizon.drawString(string, x, y, 2);

    //fill my sprite
    spr_text.setTextDatum(MC_DATUM);
    spr_text.drawString(string, 0,0,2);
    
    //This draws outside the main sprite
    //spr_text.pushSprite(x,y);
    //this draws in another dimension or something.
    spr_text.setPivot(x,y);
    spr_text.pushRotated(-roll);
  }

  spr_text.deleteSprite();

Bodmer, 2020-10-11:
See Rotated_Sprite_2 example.

Note that:
1.The pointer to the tft class is always passed to the sprite class on creation (lines 29 and 30 of example)
2. The pointer to the desintaion class sprite is passed to the source sprite that is to be rotated and pushed to the destination, see line 149

DrJaymz, 2020-10-12:
Thanks for this, I changed the constructor ref to the main tft object and then referenced the sprite I wanted to push it to and it all worked fine.  I was obviously not with it.  It now looks and works really amazing.

I was using 80MHz spi and it never gave me an issue until I started pushing super huge sprites and I have changed to DMA and reduces the clock rate and it still silky smooth.

closed 2020-10-12 by DrJaymz