Bodmer/TFT_eSPI #786: It works, but after 40 minutes the Display becomes white.

felix303, 2020-10-11:
Everthing seems to work fluently. All examples compiling and uploading without any issues.
The graphics (i.e Meters Demo) are displayed correctly. But after nearly 40 minutes my  display (Walfront ili9341 2.4") becomes white.
When I press reset button on esp32 everything is working again for the next 40 minutes when it crashes agein.
This happens with all examples. Tried it on two different esp32. Also tried with supplying 3.3v power externally to the screen.
Anyone here who used this display for more than 1 hour or experienced same behavior?
I have now ordered new displays from different brands. Maybe it is an issue of the display?!
My installed Library is "TFT_eSPI Library Version is 2.2.23"

Bodmer, 2020-10-11:
Is it powered from a PC or laptop that is putting the USB ports into a low poer state? Try powering ESP32 from a power supply independant of a computer.

felix303, 2020-10-11:
I have powered the esp32 from laptop and also with different external usb power supply and also different cables. same result.

Bodmer, 2020-10-12:
Run a basic sketch like blink or a sketch that continuously sends messages to the serial port and see if it is the ESP32 crashing.

I have run the ESP32 with a display for months without a problem so I suspect there is something different with your hardware setup.

How is the RST (reset) pin of the display being driven? What GPIO pins have you used?

Petros144, 2020-10-12:
I had the same issue long ago. There are different reasons for this problem:

1. Long cables to the display -> clock/timing signal goes bad.
2. electromagnetic interference -> happend to me when i powered my old Transformator Weller Soldering Iron on & off.
3. Reset Line of the display is triggered or Floating.
4. Breadboard connections are dogy.

I got this solved with custom PCB with short traces to the display. driving it at 80Mhz, no problem.

felix303, 2020-10-14:
Sorry for late reply but I did a lot of tests. Different cables, shorter cables , directly connecting display and esp32 without breadboard and finally including blink into the Meters demo. Led was still blinking while display again turned into white.
So esp32 seemed to work. 
This morning my new display received (HiLetGo 2.2") and I plugged it into the breadboard again.
And it is running the meters demo for almost 12 hours !!! Long story short: The Walfront Display was not working.
Sorry for taking your time. @Bodmer  and @Petros144, thank you for your help !!! @Bodmer, what great project. Thank you for this library !!!

closed 2020-10-14 by felix303