Bodmer/TFT_eSPI #788: Driver for WaveShare TFT 4" (ILI9486)

Bilbo707, 2020-10-15:
Hello Bodmer, 
first i want to thank you for the great TFT_eSPI- library. 
It works very well with Lolin D32pro and Wemos TFT2.4".

Now i want to use the library for a greater TFT, WaveShare 4" [https://www.waveshare.com/4inch-tft-touch-shield.htm](url) and ESP32 UNO board.
But i have no success. I get no display output, the display ist always dark.

I tried it with the following user_setup.h settings:

```
#define ILI9486_DRIVER
#define TFT_WIDTH  320
#define TFT_HEIGHT 480

#define TFT_CS   5    // Chip select control pin D8
#define TFT_DC   14  // Data Command control pin
#define TFT_RST  12  // Reset pin 
#define TFT_BL 13     // LED back-light 
#define TOUCH_CS 17     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY  20000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
```

Have you any idea whats wrong? 
Is TFT_eSPI library suitable for this display?

Looking forward for your answer.

Bodmer, 2020-10-15:
The display appears to have a shift register on the back so is not a normal SPI interface direct to the TFT driver. You might get lucky if you use:
```
#define RPI_DISPLAY_TYPE
```
As the first line in setup and set the SPI frequency to 20MHz.

The backlight is controled by GPIO 13 so make sure that is set in the correct state to get the backlight LEDs illuminated.

Bilbo707, 2020-10-16:
Thank you very much for the quick response and the helpful note.
Now im lucky with the bigger display.

Have a nice weekend!

Bodmer, 2020-10-16:
Good to hear. Hope your project goes well.

closed 2020-10-16 by Bodmer