Bodmer/TFT_eSPI #790: Image blinks and dissapear

Girum, 2020-10-19:
Hi! I used the library and managed the pins and driver (mine is St7735). In Adafruit Library it went ok. In this, when I reset the code, the image generated (used TFT_CLOCK example), the image blinks and return to a white screen. When I click another time in the reset pin, the image blinks again, and back to white screen. What can be?

Girum, 2020-10-19:
I can ow see using this configuration:

`#define SPI_FREQUENCY   5000000
`

The only frequency that work is 5MHz and 1MHz. But, why?

Bodmer, 2020-10-19:
Typically the ST7735 will operate at 27MHz. The display requires an adequate power supply and do not forget the 0V (GND) connection.

Without a full knowledge of your design including all interconnections I cannot advise further.

Girum, 2020-10-19:
Hi! I used this connection: [http://gogs.info/blog/2020/03/connecting-177-inch-TFT-LCD-screen-ESP8266/](url)

Some examples run at 1MHz too, with exception the JPG flash.

So the pins: 

`#define TFT_CS   PIN_D2 
#define TFT_DC   PIN_D1 
#define TFT_RST  PIN_D6 `

And the Configurations:

`#define ST7735_DRIVER  
 #define TFT_WIDTH  128
 #define TFT_HEIGHT 160
#define ST7735_BLACKTAB`

This is my screen: [https://www.arthurwiz.com/uploads/1/2/0/0/12008432/845924223_orig.jpg](url)
[https://www.arthurwiz.com/uploads/1/2/0/0/12008432/published/lcd-st7735-back.jpg?1535231696](url)

Bodmer, 2020-10-20:
Do not use pin D6 as that is the MISO pin for hardware SPI, it will be changed to an input by the SPI library and you will get spurious resets. Try another pin.

Suggested defaults are:
#define TFT_CS   PIN_D8  // Chip select
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin

Girum, 2020-10-20:
Thanks. Can close now!

closed 2020-10-20 by Girum