Bodmer/TFT_eSPI #794: Screen blink when I write long text

Girum, 2020-10-20:
I test the TFT with BLE in ESP32. When I send a long text, like 1234567891234567891234, the screen blinks. Before I use the Yeld() (like recommended). The screen blinks and erase text. Removing the Yeld(), the screen blinks but maintain the text. What causes the screen blink?

Until the last 2 digits, even the text goes to line below, it not blinks.

Code:

`#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>

#define SERVICE_UUID1    "FFFFFFF0-FFFF-FFFF-FFFF-FFFFFFFFFFFF" // UART service UUID
#define CHARACTERISTIC11 "FFFFFFF1-FFFF-FFFF-FFFF-FFFFFFFFFFFF"
#define CHARACTERISTIC12 "FFFFFFF2-FFFF-FFFF-FFFF-FFFFFFFFFFFF"

#define SERVICE_UUID2    "FFFFFF0F-FFFF-FFFF-FFFF-FFFFFFFFFFFF" // UART service UUID
#define CHARACTERISTIC21 "FFFFFF1F-FFFF-FFFF-FFFF-FFFFFFFFFFFF"
#define CHARACTERISTIC22 "FFFFFF2F-FFFF-FFFF-FFFF-FFFFFFFFFFFF"

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h
#define TFT_GREY 0x5AEB

BLEServer *pServer = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
std::string tx1;
std::string tx2;
BLECharacteristic * pRxCharacteristic1;
BLECharacteristic * pRxCharacteristic2;

//Callbacks do servidor BLE
class MyServerCallbacks: public BLEServerCallbacks {

    //Ao se conectar
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    //Ao se desconectar
    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void printTFT(String msg) {
  Serial.println(msg);
  tft.fillScreen(TFT_BLACK);
  
  // Set "cursor" at top left corner of display (0,0) and select font 2
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line is there is room for the text with tft.print)
  tft.setCursor(0, 0, 2);
  // Set the font colour to be white with a black background, set text size multiplier to 1
  tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1);
  // We can now plot text on screen using the "print" class
  tft.println(msg);
}

class CallbackUUID1: public BLECharacteristicCallbacks {
  void onWrite(BLECharacteristic *pCharacteristic) {
    std::string rxValue = pCharacteristic->getValue();
    Serial.println("********");
    Serial.print("Value written: ");
    Serial.println(rxValue.c_str());
    tx1 = rxValue.c_str();

    printTFT(String(rxValue.c_str()));

    if(String(rxValue.c_str()) == "batata"){
      Serial.println("Este foi o valor correto.");
    } else {
      Serial.println("Este não foi o valor correto.");
    }

    Serial.println("********");
  }

  void onRead(BLECharacteristic *pCharacteristic) {
    
  }
};

class CallbackUUID2: public BLECharacteristicCallbacks {
  void onWrite(BLECharacteristic *pCharacteristic) {
    std::string rxValue = pCharacteristic->getValue();
    Serial.println("********");
    Serial.print("Value written: ");
    Serial.println(rxValue.c_str());
    tx2 = rxValue.c_str();
    
    if(String(rxValue.c_str()) == "cebola"){
      Serial.println("Este foi o valor correto.");
    } else {
      Serial.println("Este não foi o valor correto.");
    }

    Serial.println("********");
  }

  void onRead(BLECharacteristic *pCharacteristic) {
    
  }
};

void setup() {
  Serial.begin(115200);

  tft.init();
  tft.setRotation(3);
  tft.fillScreen(TFT_GREY);

  //Criando um dispositivo BLE conectável
  BLEDevice::init("Teste de BLE ESP-32");

  //Criando um servidor BLE com Callbacks
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  //Criando os serviços
  BLEService *pService1 = pServer->createService(SERVICE_UUID1);
  BLEService *pService2 = pServer->createService(SERVICE_UUID2);

  BLECharacteristic * pTxCharacteristic1 = pService1->createCharacteristic(
                       CHARACTERISTIC11,
                      BLECharacteristic::PROPERTY_WRITE
                    );
  BLECharacteristic * pTxCharacteristic2 = pService2->createCharacteristic(
                       CHARACTERISTIC21,
                      BLECharacteristic::PROPERTY_WRITE
                    );
  pRxCharacteristic1 = pService1->createCharacteristic(
                       CHARACTERISTIC12,
                      BLECharacteristic::PROPERTY_NOTIFY
                    );
  pRxCharacteristic2 = pService2->createCharacteristic(
                       CHARACTERISTIC22,
                      BLECharacteristic::PROPERTY_NOTIFY
                    );
                    
  pTxCharacteristic1->setCallbacks(new CallbackUUID1());
  pTxCharacteristic2->setCallbacks(new CallbackUUID2());

  pRxCharacteristic1->setCallbacks(new CallbackUUID1());
  pRxCharacteristic2->setCallbacks(new CallbackUUID2());

  pRxCharacteristic1->addDescriptor(new BLE2902());
  pRxCharacteristic2->addDescriptor(new BLE2902());

  pService1->start();
  pService2->start();

  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
  
}

void loop() {
if (deviceConnected) {
        pRxCharacteristic1->setValue(tx1);
        pRxCharacteristic1->notify();
        
        pRxCharacteristic2->setValue(tx2);
        pRxCharacteristic2->notify();
        delay(10); // bluetooth stack will go into congestion, if too many packets are sent
  }

    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
    // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }

    //while(1) yield();
}`

Bodmer, 2020-10-20:
I do not understand the problem. Create a simpler example I can run which demonstrates the problem and only includes the TFT_eSPI library.

Girum, 2020-10-21:
Hi! This a little bit example:

`
#include <TFT_eSPI.h>
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();
#define TFT_GREY 0x5AEB

void setup() {
  Serial.begin(115200);

  tft.init();
  tft.setRotation(3);
  tft.fillScreen(TFT_BLACK);:

}

void loop() {
  tft.fillScreen(TFT_BLACK);
  
  // Set "cursor" at top left corner of display (0,0) and select font 2
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line is there is room for the text with tft.print)
  tft.setCursor(0, 50, 2);
  // Set the font colour to be white with a black background, set text size multiplier to 1
  tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1);
  // We can now plot text on screen using the "print" class
  tft.println(msg);
 delay(3000);
}`

so, I`ve tried and remove tft.fiilScreen... It resolves the blink, but... When i write text again, it overrides only the text I`ve write before (if the previous text was too long, it remains characters). I undestand that fillScreen clear the screen, but, it blinks everytime.

Bodmer, 2020-10-21:
This happens because the strings get shorter and thus do not cover up old values. One simple way around this is to pad the string with spaces so the old values are always over-written:
```
void loop() {
tft.setCursor(0, 50, 2);
tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1);
if(count < 0) {
   count = 100;
}
String msg = "Count" + String(count);
  tft.println(msg+"   ");
count --;

delay(50);
}
```
The library also provides a drawString() function, datum setting and automatic padding to a specified width using the SetTextPadding() function. See examples.

closed 2020-10-21 by Bodmer