Bodmer/TFT_eSPI #799: add samd21 specific support

hghile, 2020-10-22:
I recently started doing stuff with LCDs and this library is great.  I started my testing on esp32s with both an ST7735 and ILI9163 (though that one had RGB/BGR issues i had to flip around), and things worked smoothly.  Then I decided to go to a seeeduino xiao for the size savings, and it's less smooth.  Still works, but I want it smoother.  The samd21 chip on the seeeduino is fast enough to do the processing, but the slowdown seems to be in SPI, because it's using the generic processor code.
seeed has a fork of this library with what looks like some samd specific spi stuff,
https://github.com/Seeed-Studio/Seeed_Arduino_LCD
and it seems like it goes about 10% faster.  I'm not sure if there's more gain to be had.
The samd chip also supports DMA, and through some hackery I was able to use 
https://github.com/adafruit/Adafruit_ZeroDMA
and send data more than twice as fast.  It'd be nice if that was supported too.

Bodmer, 2020-10-22:
Good to hear about your project successes.
Adding optimised support for a processor requires a deep understanding of the hardware and associated software environment. I do not own a SAMD based board and I am not familiar with that processor so adding optimisation is not on my to do list. Despite the promotional info the SAMD21 is quite a low performance processor compared to the ESP32 and with the restricted SPI speed  and RAM size it is unlikely that a dramatic performance improvement would be possible. For physicsize an ESP8266 may be a better option for you and is supported with optimised code. If you need DMA then an ESP32 or STM32F1/3/4/7 processor is an option. For projects that do not need WiFi, I like the STM32F411 based "black pill" boards which have good performance and RAM size.

closed 2020-10-22 by Bodmer

hghile, 2020-10-23:
I couldn't find dev boards for the esp32 or 8266 that were that small.  a plain esp8266-12 might be ok, but still bigger than the xiao, and more difficult to work with.  I did find myself limited by the ram on samd21, but was able to work around it and the processing was fast enough; it was really the basic spi speed that was the issue.  But with my dma hack it was much better, maybe half the frame rate of what I was getting on an esp32, which is totally acceptable, around 25fps.  I think it is the dma stuff that makes this chip usable for applications like this.  eg, adafruit uncanny eyes (which is awesome that's it's an example in your library) runs on the samd21 based adafruit m0 boards. The dma stuff looks hacked in to uncanny eyes, but it also seems like it's part of adafruit gfx spitft class.  I suspect there'd be room in your library too.
I'd be happy to donate to get you a samd21 board.
In any case, thanks!

toetoast, 2020-10-26:
Hi Hghile, if you are looking for an ESP32 based Dev Board (that is of small size) then have a look at this MCU:

- TinyPICO

It runs at 240MHz and has very fast SPI bus over DMA. It has large onboard SRAM / FLASH capacity and runs amazingly on this Library.

I personally have this MCU and have made successfuly high-speed LCD projects with it.

Regards
Tony

Bodmer, 2020-10-27:
Hi Hghile, unfortunately I do not ave the time available to investigate, code and debug hardware optimisation for the samd21.