Bodmer/TFT_eSPI #801: TTGo T-Display drawString positioning er

vytux-com, 2020-10-23:
S

vytux-com, 2020-10-23:
Screen Setup in PIO:
build_flags = 
  -DUSER_SETUP_LOADED=1
  -DST7789_DRIVER=1
  -DTFT_WIDTH=135
  -DTFT_HEIGHT=240
  -DCGRAM_OFFSET=1
  -DTFT_MOSI=19
  -DTFT_SCLK=18
  -DTFT_CS=5
  -DTFT_DC=16
  -DTFT_RST=23
  -DTFT_BL=4
  -DTFT_BACKLIGHT_ON=HIGH
  -DLOAD_GLCD=1
  -DLOAD_FONT2=1
  -DLOAD_FONT4=1
  -DLOAD_FONT6=1
  -DLOAD_FONT7=1
  -DLOAD_FONT8=1
  -DLOAD_GFXFF=1
  -DSMOOTH_FONT=1
  -DSPI_FREQUENCY=40000000
  -DSPI_READ_FREQUENCY=6000000

Bodmer, 2020-10-25:
There was a change as the fonts now permit partial off screen rendering.

This appears to only affect displays with an offset, I will investigate further.

Thanks for reporting this.

Bodmer, 2020-10-25:
I think this has been corrected at 2.2.23.

The latests version is 2.3.2 on Github, and that one is working fine, but I think this was corrected at 2.2.23.

Please check with latest version.

vytux-com, 2020-10-26:
I double-checked again but I am still having this issue with 2.3.2

I actually checked what feels like every version between 2.3.2 and 2.2.0 when trying to find the version where it broke

Bodmer, 2020-10-27:
OK,

I used Setup25_TTGO_T_Display.h for my tests and this short sketch and it displays in the correct position.
```
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setTextSize(1);
  tft.setTextColor(TFT_WHITE,TFT_BLACK);
  tft.drawString(F(" `` OpenStill ``"),24,4,2);
  while(1);
```
Result:
[image]

Can you test the same sketch.

Bodmer, 2020-10-27:
Just checked and I am using version 2.2.23 for the above test.

vytux-com, 2020-10-28:
OK, so your code worked for me as well.

After a little more investigation I found that the following line in my Setup routine was causing this issue: `tft.setTextDatum(MC_DATUM);`

Not sure what that is supposed to do as the rest seems to be working OK without it. I got that line from TTGo example I believe.

Bodmer, 2020-10-28:
OK, that explains it, in earlier versions of the library the drawString code pulled the strings wholy onto the screen if the sketch tried to render off screen. This was to avoid character corruption of the RLE fonts. An update allowed all fonts to be partially off screen.

Hope you project now proceeds well.

closed 2020-10-28 by Bodmer