Bodmer/TFT_eSPI #802: Setup36_RPi_touch_ILI9341 not not pointing to ILI9341 driver

mfarabee, 2020-10-23:
I am new to using this library, so I apologize in advance, if I misunderstood the proper use of the setup files:

It looks like the Setup file "Setup36_RPi_touch_ILI9341.h" does not define the driver as ILI9341 (as described in the title), but instead uses the ST7796 driver definition.  I believe that either the file name should changed to reflect the driver being used or the driver definition inside the file should be changed. 

Details:
I am using a 320x240 TFT display that uses SPI, ILI9341 and touch with an ESP32 and the Arduino IDE.

I selected as a starting point the "Setup36_RPi_touch_ILI9341.h" file. I found that it did not work because the file does not define the ILI9341, but instead defined the ST7796 driver. After changing this to define the  ILI9341 driver it did function properly (I also commented out the RPI_DISPLAY_TYPE definition, since I was not using a RPI).

File contains:
#define ST7796_DRIVER // 20MHz maximum SPI

should contain:
#define ILI9341_DRIVER

Bodmer, 2020-10-25:
Thanks for pointing this out, it appears I made a temporary change that accidentally got pushed to the master repository.

I will correct this.

Bodmer, 2020-10-25:
Done

closed 2020-10-25 by Bodmer