Bodmer/TFT_eSPI #803: Sprite / bounding box issue

jlder, 2020-10-24:
Hello and thanks again for your great library.
There seem to be a problem with the bounding box when using sprites.
This happens when the sprite is at the edge of the display.
I am using an ILI9341 with an ESP32.
it can be reproduced by using the Animated_dial example and just extending NEEDLE_RADIUS to 120.
When the needle is around 210 on the background dial, the grab and paste of the area below the sprite is not done correctly and the background picture is changed.
For now I solved the problem by having 2 dots left between sprite end TFT edge.
Regards,
JL

Bodmer, 2020-10-25:
The bounding box algorithm adds a 2 pixel border to allow for rounding errors during rotations. Increasing the needle length pushes part of the bounding box off the screen. The readRect function does not handle off screen boundaries. At some point I will update readiRect to handle this, in the meantime make sure the rectangle does not move off screen.

Bodmer, 2020-10-25:
Off screen support for readRect() has now been added.

closed 2020-10-25 by Bodmer