Bodmer/TFT_eSPI #807: Feature Request - Support for ESP-S2 out of the box. I have it working but my solution is messy.

nikthefix, 2020-10-26:


Bodmer, 2020-10-27:
It is on my to do list but I will wait for the Arduino ESP32 board package to support the ESP32-S2 first.

closed 2020-10-31 by Bodmer

nikthefix, 2020-12-10:
Hi Bodmer, just in case this might help anyone experimenting with this:

[2020-12-10 09.41.25.mov.zip](https://github.com/Bodmer/TFT_eSPI/files/5671909/2020-12-10.09.41.25.mov.zip)

Here's TFT_eSPI / eFEX working on esp32-s2 with jpeg decoder and 30 frames of 150x150 24bit sprites in psram. Display is st7789 and I even had to slow the frame rate with delays as the demo was too fast!

The workarounds to get this running smoothly were few and far between and much simpler than I had previously thought. 

For me, conversion from esp32 involved only:

VSPI references to FSPI
setSwapBytes mod in eFEX.cpp
#ifdef ESP32 references to ESP32S2

Nick

YikeStone, 2021-03-19:
@nikthefix could you please share your working code of ST7701S driver for ESP32? I have a round screen and have tried different alternate boards similar to ST7701S but they dont work.

nikthefix, 2021-03-19:
Hi Rishabh,
The ST7701S driver was a feature request for Bodmer and the TFT_eSPI library. I don't have a driver myself. For circular TFT I do use GC9A01 (240x240) and that works well with ESP32 / TFT_eSPI with the included driver.

YikeStone, 2021-03-22:
Hi @nikthefix , thanks for the recommendation. I will try and get this TFT.

sbonaime, 2021-09-06:
@nikthefix Can you provide us your "messy solution" ? I want to use an ESP32-S2 with a ST7U789. 

Thanks

nikthefix, 2021-09-07:
@sbonaime My messy solution is out of date as Arduino ESP32 Core is now at 2.0.0 from the board manager - use the dev branch boards manager url in preferences: 

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

In Bodmer's User_Setup.h uncomment line 338 to  USE_HSPI_PORT as the esp32-s2 has no VSPI port reference, just HSPI and FSPI. Your S2 dev board spi pin definition defaults will probably need to be identified or changed as they are often not the same as for ESP32.

Bodmer, 2021-11-18:
The library now supports the ESP32 S2, this has been tested with the Arduino ESP32 board package 2.0.1.

A new setup file example has been added:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70_ESP32_S2_ILI9341.h