Bodmer/TFT_eSPI #809: bloody SPIFFS

bogus105, 2020-10-27:
Dear Bodmer,

i'm trying and trying to use an example from your fantastic library - "TFT_SPIFFS_BMP". I've managed to upload BMP to SPIFFS using in Arduino IDE -> Tools -> ESP8266 Sketch Data Upload. 
Then i hit compile the sketch but the process stops at line:
`bmpFS.seek(seekOffset);`
with the following error log:

---------------------------------------------------------------------------------------------------------------------------------
C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_831325\BMP_functions.ino: In function 'void drawBmp(const char*, int16_t, int16_t)':

BMP_functions:39:28: error: no matching function for call to 'fs::File::seek(uint32_t&)'

       bmpFS.seek(seekOffset);

                            ^

C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_831325\BMP_functions.ino:39:28: note: candidate is:

In file included from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_831325\TFT_SPIFFS_BMP.ino:20:0:

C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/FS.h:66:10: note: bool fs::File::seek(uint32_t, fs::SeekMode)

     bool seek(uint32_t pos, SeekMode mode);

          ^

C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/FS.h:66:10: note:   candidate expects 2 arguments, 1 provided

exit status 1
no matching function for call to 'fs::File::seek(uint32_t&)'
---------------------------------------------------------------------------------------------------------------------------------
I'm using Arduino IDE 1.8.9

Can't figure it out why it is not compiling...

Bodmer, 2020-10-27:
You appear to have a very old version (2.3.0) of the ESP8266 board package loaded. I am using 2.7.4 and see no problems with a NodeMCU 1.0 board. I have tried 2.3.0 and get the same error as you do. So you need a much more recent version where the API became aligned with the SD library calls.

Bodmer, 2020-10-31:
I assume all works now.

closed 2020-10-31 by Bodmer

bogus105, 2020-10-31:
Dear Bodmer,
i'm still at work, had no time to check it. So far i've managed to update board manager to 2.7.4 - as i'm in the remote area and internet connection here is slooooooow. It took a while:) Maybe today i'll find the time to play with SPIFFS and JPEGs. 
Thank you for reply/.

bogus105, 2020-11-01:
Yesterday i've tried to download data from sketch data folder to esp8266 using upload tool but after changing board manager from 2.3.0 to 2.7.4 it doesn't even upload data from sketch data folder to esp... Should i change this upload tool as well to be able to update SPIFFS?

Bodmer, 2020-11-01:
Yes, update the upload tool. Note also that you can not upload if the serial monitor window is open.

bogus105, 2020-11-01:
I'll try. Thanks!

bogus105, 2020-11-03:
Dear Bodmer,

all updated all works. JPGs are now successfully uploaded into SPIFFS and then successfully displayed on TFT. Now i need to figure out how to control displaying the JPGs on TFT without using their file names as i need to do something like this: " display the file number 4" no matter what is the file name, important is that it is the 4th file in SPIFFS - and here question about files organisation in SPIFFS: every time i'm uploading 'data' folder content to SPIFFS the SPIFFS is over written entirely (?) so that is why i can see no old files on the list. Are the files always organised in alphabetic order? 

The other way to solve my problem is manipulating the string so i can "build" the file name to be displayed on TFT, i.e. i'll put JPGs like this: 1.jpg, 2.jpg, 3.jpg, 4.jpg.  Then in my program i can use an integer with values 1 to 4 and if i need to display, sat 3.jpg i need to create a string variable (?) somehow and pass it to drawJpeg function. Is there some elegant and efficient way to achieve that?

Thank you for all your support:)

Bodmer, 2020-11-03:
There is an [example here](https://github.com/Bodmer/TJpg_Decoder/tree/master/examples/All_SPIFFS) that scans the SPIFFS directory to sequentially read the file names.  That example uses the TJpg_Decoder library. The key part of the sketch is:
```
  fs::Dir directory = SPIFFS.openDir("/");
  while (directory.next()) {
    String strname = directory.fileName();
    // If filename ends in .jpg then load it
    if (strname.endsWith(".jpg")) {
      loadFile(strname.c_str());
    }
  }
```
You could use the same approach and select file "N" in the list.