Bodmer/TFT_eSPI #810: Issue Rendering JPG Files

tonyc770, 2020-10-28:
I am using the ILI9488 480x320 display with NodeMCU 1.0.
The display seems to work fine with text and graphics, buy when reading jpg images from SPIFFS, the image appears corrupt.
The sketch that I am using is NodeMCU_Jpeg_ESP - JPEG_Functions.ino

In the user_setup.h file I select the ILI9488_DRIVER and follow the appropriate pin connections to the TFT and NodeMCU.

Any suggestions?

Thanks

Bodmer, 2020-10-30:
I see that example references an early library that is not deprecated. I will update the example to work with TFT_eSPI.

Bodmer, 2020-10-31:
I have updated the example, you can[ find it here](https://github.com/Bodmer/JPEGDecoder/tree/master/examples/Other%20libraries/NodeMCU_Jpeg_ESP).

closed 2020-10-31 by Bodmer