Bodmer/TFT_eSPI #811: conflict with ethernet SPI device

infrafast, 2020-10-28:
**Platform**
Hardware: ESP-12
Core Version: 2.6.3 and also 2.7
Development Env: Arduino IDE 1.5
Operating System: Windows
Settings in IDE
Module: Wemos D1 mini r2
Flash Mode:
Flash Size: 4MB
lwip Variant: v2 Lower Memory
Reset Method:
Flash Frequency: [40Mhz]
CPU Frequency: 80Mhz
Upload Using: both OTA and SERIAL
Upload Speed: 115200

**Description**

I have a TFT screen (use TFT_eSPI) that I have to use together with the Ethernet module w5500 (library beta (0.0.1)  https://d-a-v.github.io/esp8266/Arduino/index.html

Both are on SPI, TFT uses CS(D1) and Ethernet uses CS(D2), they share
D5 GPIO14 5 SCLK
D7 GPIO13 7 MOSI
and only Ethernet uses D6 MISO

TFT screen test output: 

12:25:51.784 -> [code]
12:25:51.784 -> TFT_eSPI ver = 2.2.14
12:25:51.784 -> Processor    = ESP8266
12:25:51.784 -> Frequency    = 80MHz
12:25:51.784 -> Voltage      = 3.32V
12:25:51.784 -> Transactions = No
12:25:51.784 -> Interface    = SPI
12:25:51.784 -> SPI overlap  = No
12:25:51.784 -> 
12:25:51.784 -> Display driver = 7735
12:25:51.784 -> Display width  = 128
12:25:51.784 -> Display height = 160
12:25:51.784 -> 
12:25:51.784 -> TFT_CS   = PIN_D1
12:25:51.784 -> TFT_DC   = PIN_D8
12:25:51.784 -> 
12:25:51.784 -> Display SPI frequency = 27.00
12:25:51.784 -> [/code]

**ISSUE:**

The setup works good, the TFT is properly initiated and the W5500 module is assigned an IP address correctly
However as soon as we call a function that engage the SPI bus (in my case the TFT screen), looks like the Ethernet module loose its IP address and the link turn to OFF. It continues to work as you can still call HardwarStatus and check Link function and the TFT is also displaying properly. I suspect an implementation issue either in the  ethernet library code or in the TFT library regarding switching from one device to another on the SPI bus, looks like the TFT keep the bus for him...

code below:

        #include <SPI.h>
	#include <Ethernet.h>
	#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip

	byte mac[] = {0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02};

	long loopElapsedTime = 0;

	TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

	void setup() {
	  Serial.begin(115200);
	  while (!Serial);
	  tft.init(); tft.fillScreen(TFT_BLACK); tft.setTextColor(TFT_WHITE,TFT_BLACK);
	  
	  loopElapsedTime = millis();
	  
	  Ethernet.init(D2);  // ESP8266 with Adafruit Featherwing Ethernet

	  Serial.print("\nCode before TFT call. Initialize Ethernet with DHCP: ");
	  if (Ethernet.begin(mac) == 0) {
		Serial.println("Failed to configure Ethernet using DHCP");
		while (1); // do not continue
	  }
	  Serial.println(Ethernet.localIP());      // at that point in time we have an IP address and linkstatus is LinkON
	  Serial.printf(" link: %i",Ethernet.linkStatus()); 
	}

	void loop() {  
	  if ((millis() - loopElapsedTime) > 1000){
		loopElapsedTime = millis();
		
		Ethernet.maintain();    // maintain does no change anything to the issue
		
		tft.setCursor(0, 0, 2);
		tft.print(loopElapsedTime); // as soon as we call a tft  function that activate the SPI device, the Ethernet driver turn to LinkOFF and unset IP address
									// if you comment this call, then all works fine
		

		Serial.printf("\nCode after TFT call. link: %i",Ethernet.linkStatus()); 
		Serial.print("\nIP: ");
		Serial.print(Ethernet.localIP());    
	  }
	}

### Debug Messages

Here is what you get on serial line: 

12:45:54.326 -> Code before TFT call. Initialize Ethernet with DHCP: 192.168.0.211
12:45:55.017 ->  link: 1
12:45:55.345 -> Code after TFT call. link: 2
12:45:55.345 -> IP: (IP unset)

Bodmer, 2020-10-29:
Enable transaction support in the TFT_eSPI setup file.

infrafast, 2020-10-29:
Thank you Bodmer, as a first feedback looks like it solve (partially) my issue, at least now the TFT and ethernet works togetehr but the WIFI seems to be impacted. I will further test and keep you posted. Please keep the issue open for a while so I can.perform more extensive test

Bodmer, 2020-10-29:
I dubt that TFT_eSPI is affecting the Wifi, but I will keep this open.

Bodmer, 2020-11-02:
I am assuming you have solved this and it is not an issue with TFT_eSPI.

closed 2020-11-02 by Bodmer

infrafast, 2020-11-02:
Hello   the test I did so far do not reveal any issue with the library so I think we can indeed consider this issue as solved. Thanks again for having taken the time to support;