Bodmer/TFT_eSPI #813: Custom font weird behaviour

LightningLion, 2020-10-31:
Good day!

I have a ST7789 display, 1.3" and 240x240 pixels and I'm trying to use custom fonts with it.
I'm using this font:
https://aurekfonts.github.io/?font=AlbansBane

Which I convert to .ttf with an online tool.
And then convert it to Adafruit's GFX style with this tool (80 point size):
https://rop.nl/truetype2gfx/

I include the fonts locally on the sketch.

In my setup() I have the following code:
```
    tft.begin();     // initialize a ST7789 chip
    tft.setSwapBytes(true); // Swap the byte order for pushImage() - corrects endianness
    
    tft.fillScreen(TFT_BLACK);

    //tft.setTextDatum(MC_DATUM);

    tft.setFreeFont(Mando);
    tft.setTextColor(TFT_RED, TFT_BLACK);
    
    tft.print("SAS");
    delay(2000);
    tft.fillScreen(TFT_BLACK);
    tft.drawString("SAS", 0, 0);
```

The result of that code is, first, the correct text but cut in the lower part because it's not centered vertically.
Second, drawString only gets one S character, in the bottom of the screen, showing only the top half as it's cut.

Why does that happen? Also, if I generate a 100point GFX file, some characters (like the S) appear as an almost solid red block (there are some black pixels in between) but others look correct (like the A).

Is there any steps I'm missing? I know the TFT is wired correctly because I have used the library to draw bitmaps with full success. It's only with fonts that I've encountered the problem.

Bodmer, 2020-10-31:
The Adafruit font has a heigh limit due to the use of an 8 bit signed value for the bitmap offset. Ultimiately this means that fonts above about 80 points are not encoded properly and thus cannot be rendered correctly. This is a constraint imposed by the Adafruit format and not the TFT_eSPI library.

If you Mando font is small then I will need to have a copy of the font file to identify the cause of you problems.

LightningLion, 2020-11-01:
Hi, thanks for your reply. Font was indeed over 80points. The weird this is it only afects some characters.

Bodmer, 2020-11-01:
Yes, it affects the "tall" characters where the bitmap offset (last column in the struct) wraps. Here is the FreeSans font at 88 pts, the dollar character should have a negative offset of -132 but the value has wrapped in the signed 8 bit value to a positive value (124), so it no longer renders correctly:
```
const GFXglyph FreeSans88pt7bGlyphs[] PROGMEM = {
  {     0,   1,   1,  48,    0,    0 },   // 0x20 ' '
  {     1,  15, 126,  48,   21, -125 },   // 0x21 '!'
  {   238,  44,  42,  61,    9, -121 },   // 0x22 '"'
  {   469,  91, 123,  96,    2, -119 },   // 0x23 '#'
  {  1869,  83, 155,  96,    6,  124 },   // 0x24 '$'  <<<<< 124 should be a negative value -132
  {  3478, 143, 125, 153,    5, -121 },   // 0x25 '%'
  {  5713, 101, 126, 115,    9, -121 },   // 0x26 '&'
```

closed 2020-11-01 by Bodmer