Bodmer/TFT_eSPI #814: SPI.h compilation error

offgabriele, 2020-11-02:
### Issue
The compilator exits with code 1 after an error in the SPI.h header file is detected. I tried looking at  the line indicated but everything looks okay. In my sketch I don't directly use the library and the error is thrown from inside the TFT_eSPI library.
### Error
```
In file included from C:\Users\gabri\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:28:0,
                 from D:\Gabriele\Projects\Lovebox\firmware\firmware.ino:10:
C:\Users\gabri\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src/SPI.h:28:1: error: expected ',' or ';' before 'class'
 class SPISettings
 ^
```
### Hardware:
Board: ESP32 Dev Module
Core Installation version: 1.0.4
IDE name: Arduino IDE
Flash Frequency: 80Mhz
Upload Speed: 921600
Computer OS: Windows 10?

Pablo2048, 2020-11-02:
You probably forgot ';' somewhere in your code...

offgabriele, 2020-11-02:
Yes, that appears to be the error. However te error is thrown from a file I did not edit, so how could it be possible?

offgabriele, 2020-11-02:
> Yes, that appears to be the error. However te error is thrown from a file I did not edit, so how could it be possible?

> You probably forgot ';' somewhere in your code...

I checked my code and i could not find any syntax error.

luebbe, 2020-11-02:
looking at your source, you included SPI.h twice. Maybe that's the reason.

Pablo2048, 2020-11-02:
> Yes, that appears to be the error. However te error is thrown from a file I did not edit, so how could it be possible?

Its in the header so (probably) the header right before this one is the source of the problem...

Pablo2048, 2020-11-02:
... probably your `version.h` is the source? https://github.com/offgabriele/lovebox/blob/780ba84df549f310a088487007297e0c6ef32a57/firmware/version.h#L1 try to use text without () ;-)

Pablo2048, 2020-11-02:
Anyway @luebbe is also right - your code is a mess - multiple includes of the same library (SPI, TFT_eSPI).

offgabriele, 2020-11-02:
Thanks guys, the problem was with the new version of the header "version.h" as I forgot the ';' at the end of the line, turns out you were right all along @Pablo2048 , I'm sorry to have wasted your time but very thankful for your help. I really should clean up my code and i will.

closed 2020-11-02 by offgabriele