Bodmer/TFT_eSPI #817: Examples question

luebbe, 2020-11-03:
Hi Bodmer,

I played around with you library and noticed that several examples could be made generic by replacing the fixed sizes (160,320,480) with TFT_WIDTH and TFT_HEIGHT. This would get rid of some duplicate code in the examples. 

Would you accept PRs for this?

Bodmer, 2020-11-03:
This is on my todo list and a new set of replacement examples is in development. The plan is to rationalise the examples. The examples have been pulled from earlier libraries that were display size specific. As the plan is to rationalise the examples in conjunction with a user manual, a pull request would end up being superceded  and thus be nugatory effort. Thanks for the offer of support.

closed 2020-11-03 by Bodmer

luebbe, 2020-11-04:
Ok, fine by me.
Here's a reworked generic version of game of life with color coding by cell age if you like. Only tested on ST7789 240x240. I have no other device for this library.

[Cellular_Automata.zip](https://github.com/Bodmer/TFT_eSPI/files/5486138/Cellular_Automata.zip)