Bodmer/TFT_eSPI #820: Function going loop when using IRQ pin in Touch Screen

baniaque, 2020-11-07:
Trying to use interrupt routine to reading touch screen only when is pressed. Used part of test code from examples. But when I touch screen once, function is going loop and cannot be stopped even if use detachInterrupt in code. When delete tft.getTouchRaw(&x, &y) and  tft.getTouchRawZ()) form code, interrupt routine working great.

void setup() {
attachInterrupt(digitalPinToInterrupt(touch_irq_pin), touch_ISR, LOW);
}

void touch_ISR() {
  detachInterrupt(touch_irq_pin);
  uint16_t x, y;
  tft.getTouchRaw(&x, &y);
  Serial.printf("x: %i     ", x);
  Serial.printf("y: %i     ", y);
  Serial.printf("z: %i \n", tft.getTouchRawZ());
  attachInterrupt(digitalPinToInterrupt(touch_irq_pin), touch_ISR, LOW);
}

Pablo2048, 2020-11-07:
Try to use FALLING instead of LOW trigger.

baniaque, 2020-11-07:
My first choice was FALLING, exactly same behavior.

Lebo77, 2020-11-07:
I do a software denounce on the IRQ line. I disable triggering additional
reads for 100mm or there about after recieving an interrupt.

Pablo2048, 2020-11-07:
> My first choice was FALLING, exactly same behavior.

So it seems like missing pull-up resistor on IRQ pin then...

baniaque, 2020-11-07:
@Pablo2048 - but when I replace code inside touch_ISR() function to code without using tft.getTouchRaw and tft.getTouchRawZ then irq routine working great, I press screen and function running once (with FALLING) and running in loop until I relase finger from screen (with LOW).

Pablo2048, 2020-11-07:
But this is normal behavior IMHO of the XPT2046 (if you are using this chip). Did you read the documentation about the PENIRQ (page 24-25 of the datasheet) pin? Also you can get inspiration from existing library about how to handle it correctly ( https://github.com/PaulStoffregen/XPT2046_Touchscreen )

Bodmer, 2020-11-07:
I cannot offer more advice than Pablo2028, take note of the comments on the page he referenced.

As this is not an issue with the library and you have a good answer, I will close.

closed 2020-11-07 by Bodmer