Bodmer/TFT_eSPI #821: Support for WaveShare 4” 480x320 RPI + Teensy 4.0?

1ecu, 2020-11-09:
Even if it’s probably not a high priority issue with this library I’m nevertheless looking for help to get TFT_eSPI working with Teensy 4.0 + WaveShare TFT 4” 480x320 RPI (ILI9486, XPT2046 / SKU 13587 / https://www.waveshare.com/wiki/4inch_TFT_Touch_Shield). 
Touch is currently ok and display works... well, "almost":

[image]

Out of the box no example compiled but gave lots of errors about missing definition for tft_write_16().
 
So I put a custom #define TEENSY40 in User_Setup and in TFT_eSPI_Generic.h I copied some definitions from below #if  defined (RPI_DISPLAY_TYPE) to form a custom conditional block:

#if defined (TEENSY40) 
 #define tft_Write_8(C)    spi.transfer(0); spi.transfer(C) 
      #define tft_Write_16(C)  tft_Write_8((uint8_t)((C)>>8));tft_Write_8((uint8_t)((C)>>0))
      #define tft_Write_16S(C) tft_Write_8((uint8_t)((C)>>0));tft_Write_8((uint8_t)((C)>>8))
#endif

This forced definitions for tft_write_16() and they even compiled. But evidently those - or some other line(s) in somewhere - are not totally inline with the system because screen is mostly blue and fonts are blurred and skewed. Does anyone know how to fix the files properly in order to get this screen working with Teensy 4?

[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/5510998/User_Setup.zip)

Bodmer, 2020-11-09:
The RPi screens are unusual in that they need 16 bits for a command. Your change does this but the you send 32 bits where a 16 bit transfer is needed. Try this:
```
#if defined (TEENSY40)
#define tft_Write_8(C) spi.transfer(0); spi.transfer(C)
#define tft_Write_16(C) spi.transfer((uint8_t)((C)>>8));spi.transfer((uint8_t)((C)>>0))
#define tft_Write_16S(C) spi.transfer((uint8_t)((C)>>0));spi.transfer((uint8_t)((C)>>8))
#endif
```

1ecu, 2020-11-09:
Many thanks for quick reply.

I tested your suggestion but "my" definitions appear still to be "better"... ;-)

[image]

And just to make clear, the definitions were located in TFT_eSPI_Generic.h like show below and tested without conditions 
 .  .  .
 #endif // RPI_DISPLAY_TYPE

      // #define tft_Write_8(C)     spi.transfer(0); spi.transfer(C)
      // #define tft_Write_16(C)  tft_Write_8((uint8_t)((C)>>8));tft_Write_8((uint8_t)((C)>>0))
      // #define tft_Write_16S(C) tft_Write_8((uint8_t)((C)>>0));tft_Write_8((uint8_t)((C)>>8))

	#define tft_Write_8(C) spi.transfer(0); spi.transfer(C)
	#define tft_Write_16(C) spi.transfer((uint8_t)((C)>>8));spi.transfer((uint8_t)((C)>>0))
	#define tft_Write_16S(C) spi.transfer((uint8_t)((C)>>0));spi.transfer((uint8_t)((C)>>8))

  #define tft_Write_32(C) .  .  .
  .  .  . 

Should in this case the latter definitions for ftf_write_32() functions be changed also in some way?

Bodmer, 2020-11-09:
Yes, the tft_write32 functions were correct in your original version. But the 16 bit ones were not.

1ecu, 2020-11-09:
Do the compiler warnings about progmem have anything to do with this? I mean particularly with Teensy.

In file included from C:\Users\SYS_AD~1\AppData\Local\Temp\arduino_modified_sketch_80572\TFT_ring_meter.ino:20:0:
C:\Users\sys_admin\Documents\Arduino\libraries\TFT_eSPI-master/TFT_eSPI.h:47:0: warning: "PROGMEM" redefined
   #define PROGMEM
 
In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/WProgram.h:41:0,
                 from C:\Users\SYS_AD~1\AppData\Local\Temp\arduino_build_832705\pch\Arduino.h:6:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/avr/pgmspace.h:30:0: note: this is the location of the previous definition
 #define PROGMEM __attribute__((section(".progmem")))

In file included from C:\Users\sys_admin\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:17:0:
C:\Users\sys_admin\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.h:47:0: warning: "PROGMEM" redefined
   #define PROGMEM
 

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/WProgram.h:41:0,
                 from C:\Users\SYS_AD~1\AppData\Local\Temp\arduino_build_832705/pch/Arduino.h:6,
                 from C:\Users\sys_admin\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.h:30,
                 from C:\Users\sys_admin\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:17:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/avr/pgmspace.h:30:0: note: this is the location of the previous definition  #define PROGMEM __attribute__((section(".progmem")))

Bodmer, 2020-11-09:
The PROGMEM compiler warnings are not part of the SPI communication problem. Line 47 is required for the STM32 processors but it looks like the teeny board package defines it. I have never tried the library with a teensy so I am not surprised you are getting warnings.

The simplest solution in the short term is to make the change you have already made.

Bodmer, 2020-11-09:
I think this is the code you need:
```
    #define tft_Write_8(C)   spi.transfer(0); spi.transfer(C)
    #define tft_Write_16(C)  spi.transfer((uint8_t)((C)>>8));spi.transfer((uint8_t)((C)>>0))
    #define tft_Write_16S(C) spi.transfer((uint8_t)((C)>>0));spi.transfer((uint8_t)((C)>>8))

    #define tft_Write_32(C) \
      tft_Write_16((uint16_t) ((C)>>16)); \
      tft_Write_16((uint16_t) ((C)>>0))

    #define tft_Write_32C(C,D) \
      spi.transfer(0); spi.transfer((C)>>8); \
      spi.transfer(0); spi.transfer((C)>>0); \
      spi.transfer(0); spi.transfer((D)>>8); \
      spi.transfer(0); spi.transfer((D)>>0)

    #define tft_Write_32D(C) \
      spi.transfer(0); spi.transfer((C)>>8); \
      spi.transfer(0); spi.transfer((C)>>0); \
      spi.transfer(0); spi.transfer((C)>>8); \
      spi.transfer(0); spi.transfer((C)>>0)
```

1ecu, 2020-11-09:
Ok, now I must admit that your solution is better than mine :D
 
[image]

New functions did the trick and examples compile and work perfectly. Thank you so much! 

BTW, is the new code just this WaveShare tft related or has it something to with Teensy?

Bodmer, 2020-11-10:
I have updated the library with the new code.

Good to see it is working for you!

closed 2020-11-10 by Bodmer