Bodmer/TFT_eSPI #824: TFT_eSPI no touch response by WeatherAppArduiTouch

eleleljo, 2020-11-12:
Hello
I also get no response of my 2.4" touch display by using ESP32 DEV Module
I use this WeatherAppArduiTouch https://github.com/Slaveche90/WeatherAppArduiTouch
I can load all the TFT_eSPI 320x240 examples.
I get no response of the touch screen when I touch the Numbers.

In the User_Setup.h I wrote like the e-Book says:

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS 5
#define TFT_DC 4
#define TFT_RST 22
//#define TFT_RST -1
//#define TFT_BL 32
#define TOUCH_CS 14
//#define TFT_WR 22

this is the output of the Test_touch_controller

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
x: 0 y: 3286 z: 19
x: 0 y: 3484 z: 21
x: 0 y: 3181 z: 25
x: 205 y: 3138 z: 5
x: 0 y: 3499 z: 20
x: 0 y: 3502 z: 23
x: 0 y: 2392 z: 15
x: 257 y: 2908 z: 25
x: 0 y: 3341 z: 29
x: 480 y: 3048 z: 4
x: 0 y: 2714 z: 23
x: 0 y: 3476 z: 21
x: 80 y: 3183 z: 20
x: 268 y: 2886 z: 19
x: 112 y: 3595 z: 22
x: 428 y: 3125 z: 29
x: 41 y: 3263 z: 34
x: 0 y: 3458 z: 5
x: 0 y: 3483 z: 22
x: 585 y: 2886 z: 27
x: 50 y: 3190 z: 33
x: 560 y: 2884 z: 23
x: 368 y: 3086 z: 5
x: 0 y: 2448 z: 14
x: 0 y: 3448 z: 27
x: 0 y: 3479 z: 27
x: 0 y: 3469 z: 23
x: 0 y: 2375 z: 16
x: 0 y: 3424 z: 21
x: 0 y: 3511 z: 20

x: 394 y: 3152 z: 21
x: 0 y: 3477 z: 22
x: 0 y: 2402 z: 15
x: 147 y: 3611 z: 24
x: 0 y: 3351 z: 7
x: 0 y: 3519 z: 24
x: 0 y: 2361 z: 16
x: 0 y: 3472 z: 24
x: 0 y: 3436 z: 13
x: 702 y: 2740 z: 19
x: 0 y: 3456 z: 17
x: 0 y: 2672 z: 19
x: 0 y: 2368 z: 13
x: 480 y: 3056 z: 4
x: 4 y: 3488 z: 22
x: 0 y: 3464 z: 24
x: 80 y: 3260 z: 37
x: 0 y: 2357 z: 11
x: 0 y: 3450 z: 5
x: 356 y: 3101 z: 5
x: 77 y: 3247 z: 37
x: 0 y: 3429 z: 21
x: 0 y: 3376 z: 29
x: 0 y: 3515 z: 21
x: 0 y: 3515 z: 20
x: 0 y: 3462 z: 24
x: 0 y: 3351 z: 9
x: 555 y: 2937 z: 27
x: 0 y: 3448 z: 27
x: 128 y: 3591 z: 24
x: 0 y: 3480 z: 20
x: 0 y: 3359 z: 18
x: 1 y: 3484 z: 20
x: 384 y: 3123 z: 5
x: 0 y: 2408 z: 11
x: 434 y: 3107 z: 37
x: 0 y: 3480 z: 20
x: 2 y: 3494 z: 32
x: 0 y: 3520 z: 15
x: 0 y: 3460 z: 69
x: 662 y: 2597 z: 13
x: 482 y: 3050 z: 4

Dos someone has an idea what is wrong at my device?

Greetings
Josef

Bodmer, 2020-11-12:
The WeatherAppArduiTouch sketch does not use the touch screen. The sketch name is misleading, it just uses the AdruiTouch case.

eleleljo, 2020-11-12:
Thank you for your professional answer. So I do not need to search further for problems. 
I recognized that the IRQ is connected at the board. If I would disconnect the IRQ, would the touch function?
What could I do, that the touch would function? I do not need the weather app. I would like to use the case and the ESP32 to measure the volume of tanks filled with fluid candle wax.

Greetings
Josef

Bodmer, 2020-11-12:
When you run the Test_touch_controller example does the Z value increase significantly and does the x, and y change to steady values?

Bodmer, 2020-11-12:
Re. IRQ, the library does not use that pin so it does not matter if it is connected or not.

eleleljo, 2020-11-12:
Thanks for your response.

When you run the Test_touch_controller example does the Z value increase significantly and does the x, and y change to steady values?
this is the result, when I made two diagonals
from upper left tu the down rightside 
and fron the left downside to the right upper side.
x: 0     y: 3360     z: 19 
x: 0     y: 3490     z: 19 
x: 0     y: 3475     z: 16 
x: 64     y: 3624     z: 30 
x: 0     y: 3518     z: 21 
x: 334     y: 3160     z: 3 
x: 347     y: 3114     z: 23 
x: 397     y: 3114     z: 3 
x: 331     y: 3146     z: 3 
x: 0     y: 3068     z: 26 
x: 629     y: 410     z: 2068 
x: 1038     y: 736     z: 2117 
x: 1502     y: 1436     z: 2017 
x: 2150     y: 2248     z: 1769 
x: 3062     y: 3041     z: 1778 
x: 3579     y: 3612     z: 1662 
x: 83     y: 3169     z: 3 
x: 0     y: 3512     z: 22 
x: 0     y: 3482     z: 19 
x: 0     y: 3520     z: 23 
x: 0     y: 3423     z: 2 
x: 2669     y: 1359     z: 2218 
x: 1506     y: 2390     z: 1645 
x: 728     y: 3056     z: 1119 
x: 106     y: 3288     z: 27 
x: 347     y: 3120     z: 22 
x: 0     y: 3456     z: 11 
x: 387     y: 3123     z: 3 
x: 0     y: 3446     z: 4 
x: 57     y: 3183     z: 11 
x: 0     y: 2600     z: 28 
x: 0     y: 3357     z: 21 
x: 329     y: 3115     z: 27 
x: 0     y: 3508     z: 19 
x: 120     y: 3162     z: 3 
x: 0     y: 3526     z: 30 
x: 179     y: 3132     z: 3 

Thanks fpor helping me

Bodmer, 2020-11-13:
Looks like the touch screen is responding.

Have you tried the "Touch_calibrate." sketch?

eleleljo, 2020-11-13:
Thank you, it works now

Bodmer, 2020-11-13:
Great!

closed 2020-11-13 by Bodmer

eleleljo, 2020-11-13:
Thank you for your help,
The sketch Keypad_240x320 always overjumped the touch_calibrate();
I saw always immediately the buttons. As I pressed once the reset button, the calibrating corners appeared. After calibrating the touch functioned.

Thanks a lot for your help