Bodmer/TFT_eSPI #825: Fastest 3.5" DMA display

Guerillah, 2020-11-12:
Hey Bodmer,

Can I get a recommendation from you on the fastest 3.5" DMA display that your library supports? I'll be implementing LVGL along with your library and I want it to be fluid as possible. Thank you.

Bodmer, 2020-11-12:
You do not say what processors or resolution you will be using...

SPI screen typically have a an operating clock rate in the range 16-30MHz, so using them above that rate may lead to problem. However ILI9341 an ST7796 screens operarate reliably at higher frequencies. The IL9341 can operate at 55MHz reliably when driven by an STM32 processor, or 40MHz from an ESP32. The ST7798 appears to work reliably at 80MHz with an ESP32 but your mileage may vary.

I ususally use the "boing ball" example for DMA performance test. Here are the results for various processors and SPI clock rates:
```
// STM32F767 55MHz SPI 170 fps without DMA
// STM32F767 55MHz SPI 227 fps with DMA
// STM32F446 55MHz SPI 110 fps without DMA
// STM32F446 55MHz SPI 187 fps with DMA
// STM32F401 55MHz SPI  56 fps without DMA
// STM32F401 55MHz SPI 120 fps with DMA

// STM32F767 27MHz SPI  99 fps without DMA
// STM32F767 27MHz SPI 120 fps with DMA
// STM32F446 27MHz SPI  73 fps without DMA
// STM32F446 27MHz SPI  97 fps with DMA
// STM32F401 27MHz SPI  51 fps without DMA
// STM32F401 27MHz SPI  90 fps with DMA

// Blue Pill - 36MHz SPI *no* DMA 36 fps
// Blue Pill - 36MHz SPI with DMA 67 fps
// Blue Pill overclocked to 128MHz *no* DMA - 32MHz SPI  64 fps
// Blue Pill overclocked to 128MHz with DMA - 32MHz SPI 116 fps

// ESP32     - 8 bit parallel     110 fps (no DMA)
// ESP32     - 40MHz SPI *no* DMA  93 fps
// ESP32     - 40MHz SPI with DMA 112 fps
// ESP32     - 80MHz SPI with DMA 142 fps
```

Bodmer, 2020-11-12:
Corrected typo for controller - should have read ST7796

Guerillah, 2020-11-12:
Hey Bodmer,

It would make sense to include that information wouldn't it... thanks for your patience. 
ESP32 @ 240 Mhz, 480x320 resolution.

What resolution display was used for the benchmarks above?

Bodmer, 2020-11-12:
The boing ball runs on any size screen. As the "ball" is small it runs on any size screen.  The DMA examples can be run without a screen to evaluate performance as the fps is sent to the serial port and the processor does not rely on a screen being connected.

closed 2020-11-13 by Bodmer

Guerillah, 2020-11-17:
Bodmer,

In doing more research on the topic, I saw in some older posts that you attributed the speed of the MHS 4” Display B to its use of a high speed serial to parallel IC (unbranded / unmarked what kind). Is this the only reason it can run at 80 MHz or can I expect to get 80 MHz straight from using an ESP32 and a ST7796s display controller?

Bodmer, 2020-11-17:
The MHS 4.0 display is rated at 125MHz maximum so is the only display that is designed to run at high clock rates. The disadvantage is that it requires 16 bit command so for low pixel level transactions to different creen locations it would be slower than the ST7796. Both displays are good options for use at 80MHz. You may find the MHS 4.0 B display is easier to find online.