Bodmer/TFT_eSPI #826: ESP8266WebServer.h incompatibility?

chrisgimson, 2020-11-15:
Does anyone have any insight to a problem when combining this library with a webserver?  

The issue appears immediately on adding #include <ESP8266WebServer.h> to even a simple example sketch TFT_Print_Test.ino with absolutely no change to the actual sketch code.  The sketch will not compile and returns the error messages ..."expected '}' at end of input

[image]

Bodmer, 2020-11-15:
I get the same error, but if you put the header load first all is well:
```
#include <ESP8266WebServer.h>

#include <TFT_eSPI.h> // Graphics and font library for ILI9341 driver chip
#include <SPI.h>

#define TFT_GREY 0x5AEB // New colour

TFT_eSPI tft = TFT_eSPI();  // Invoke library

etc. etc

```

chrisgimson, 2020-11-15:
Great.  I will resume my application and feedback whether it works OK or if I still have a problem.
... Chris

chrisgimson, 2020-11-18:
I am happy to feedback that after relocating the #include <ESP8266WebServer.h> line to the start of the code, the local webserver and access point webservers work fine so clearing the incompatibility issue.  Can I suggest you add a line in your into text highlighting the need if using that library.  Many thanks for your initial fast response - problem sorted :)