Bodmer/TFT_eSPI #827: Arduino due

EliasA97, 2020-11-17:
This library doesn't compile for arduino due even if it's a 32bit MCU.

Bodmer, 2020-11-17:
This is down to a function ambiguity and the way "true" is defined for the DUE.

Should now be fixed in latest copy on Github.

closed 2020-11-17 by Bodmer

EliasA97, 2020-11-17:
Hi Bodmer and thank's for giving your time to fix the bugs. I just downloaded the latest version of the library and tried an example but again I cant compile when I select the Due board. I get these errors:

`
In file included from C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:4554:0:

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp: In member function 'bool TFT_eSprite::pushSprite(int32_t, int32_t, int32_t, int32_t, int32_t, int32_t)':

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:834:67: error: call of overloaded 'pushImage(int32_t&, int32_t&, int32_t&, int32_t&, uint8_t*, int)' is ambiguous

       _tft->pushImage(tx, ty, sw, sh, _img8 + _iwidth * _ys, true );

                                                                   ^

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:834:67: note: candidates are:

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:1361:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, bool, uint16_t*)

 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, bool bpp8,  uint16_t *cmap)

      ^

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:1504:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, uint8_t, bool, uint16_t*)

 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, uint8_t transp, bool bpp8, uint16_t *cmap)

      ^

In file included from C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:4554:0:

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:837:76: error: call of overloaded 'pushImage(int32_t&, int32_t, int32_t&, int, uint8_t*, int)' is ambiguous

       _tft->pushImage(tx, ty++, sw, 1, _img8 + _xs + _iwidth * _ys++, true );

                                                                            ^

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:837:76: note: candidates are:

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:1361:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, bool, uint16_t*)

 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, bool bpp8,  uint16_t *cmap)

      ^

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:1504:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, uint8_t, bool, uint16_t*)

 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, uint8_t transp, bool bpp8, uint16_t *cmap)

      ^

In file included from C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:4554:0:

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:873:73: error: call of overloaded 'pushImage(int32_t&, int32_t&, int32_t&, int32_t&, uint8_t*, int)' is ambiguous

       _tft->pushImage(tx, ty, sw, sh, _img8 + (_iwidth>>3) * _ys, false );

                                                                         ^

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp:873:73: note: candidates are:

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:1361:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, bool, uint16_t*)

 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, bool bpp8,  uint16_t *cmap)

      ^

C:\Users\anast\OneDrive\�������\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:1504:6: note: void TFT_eSPI::pushImage(int32_t, int32_t, int32_t, int32_t, uint8_t*, uint8_t, bool, uint16_t*)

 void TFT_eSPI::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t *data, uint8_t transp, bool bpp8, uint16_t *cmap)

      ^

exit status 1

Σφάλμα μεταγλώττισης για την πλακέτα Arduino Due (Programming Port).

Αυτή η αναφορά θα παρείχε περισσότερες πληροφορίες με 
"Εμφάνιση πλεονάζουσα έξοδος κατά τη σύνταξη"
Επιλογή από το Αρχείο -> Προτιμήσεις.
`

Bodmer, 2020-11-17:
If you use the Arduino IDE library manager then you will not get the latest version that is on Github as it has no been released. You have to download the zip file from Github.

Alternatively just add (bool) to the lines 834, 837 and 873 in error within the Sprite.cpp file, example:
```
_tft->pushImage(tx, ty, sw, sh, _img8 + _iwidth * _ys, (bool) true );
```

EliasA97, 2020-11-17:
Well you're right. I just tested and it works. Thank's for all your effort!

Bodmer, 2020-11-17:
Great, thanks for reporting this issue.

commit 78804815 (2020-11-17): Fix #827 (function ambiguity with Arduino DUE)