Bodmer/TFT_eSPI #828: Touch Not working on ILI9488

tonyc770, 2020-11-17:
I am unable to get any of the example Touch sketches working with my ESP32 with the ILI9488 and I have tried all of them.
The text and graphics work fine.

Here is my user_setup.h mods
#define ILI9488_DRIVER 

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

I have the TFT_MISO pin GPIO19 connected to T_DO and TOUCH_CS 21 connected to T_CS and TFT_SCLK connected to T_CLK

When running the sketches I get no response from any TFT Touch and the Calibration routines don't seem to work either.

Thanks,
Tony

Bodmer, 2020-11-17:
The ILI9488 is a bit odd in that it does not tristate the data out line. Therefore do not connect the processor MISO line to the TFT, only connect it to the touch controller data out pin.

tonyc770, 2020-11-17:
That is the way I have it connected, just to the touch DO.

Bodmer, 2020-11-17:
Run the "Test_Touch_Controller", this outputs readings from the screen to the serial port.

Check your screen has a XPT2046 touch controller fitted and the touch film fitted to the front of the screen,

tonyc770, 2020-11-17:
I did run that test program. The serial output gave zero x and y data. The touchpad is attached. I'll have to check the  the controller chip.

tonyc770, 2020-11-18:
I checked the Touch controller and it is XPT2014.
So any other reason not working? BTW, I have 2 of these displays, and neither one seem to response to the Touch code.

Thanks...

Bodmer, 2020-11-18:
Do you have TFT_MOSI pin 23 connected to the touch T_DIN?

tonyc770, 2020-11-19:
OMG! for some reason I thought only the MISO needed to be connected, but both SPI data IO need to be connected, but only the MISO to the Touch controller as indicated previously!!

Thanks!!!

Bodmer, 2020-11-19:
Glad it is working now!

closed 2020-11-19 by Bodmer